MaslowCNC/GroundControl #6: Hold coordinates in GC to resume after disconnect

MakesmithTech, 2015-02-04:
"Greetings all! I love my CNC mill but it is driving me up the wall with this issue about disconnects. I had hoped that the reconnect feature would solve this, it's kinda frustrating to bump the table and have an hour + of cut time go down the drain.

Is there any way at all to store the current coordinates so that when I unplug my CNC mill and plug it back in (after restarting ground control), that the most recent coordinates are saved? This must be doable in ground control, even if not in the CNC software itself."

MakesmithTech, 2015-02-04:
Maybe the solution is to store the machine's coordinates only when the servos disconnect at the end of the run in the machine's eeprom.

MakesmithTech, 2015-02-26:
The plan:

When the servos disconnect, after being idle the machine will store it's position into EEPROM, every time the machine boots, it will load it's position from EEPROM.

On the GC side, if the machine disconnects partway through a run, GC will store the current gcode line in the settings file and pause the gcode run. When the machine reconnects or when GC is restarted the program can be resumed.

MakesmithTech, 2015-03-01:
Firmware side complete.

BarbourSmith, 2016-10-17:
This behavior has been in place for a while now, should have closed this

closed 2016-10-17 by BarbourSmith