MaslowCNC/GroundControl #7: Change Feedrate on the fly

MakesmithTech, 2015-02-04:
"I'd like to be able to scale the F values on the fly so I can slow down or increase the rate of advance while milling." -Tobor

mrfugu, 2018-08-23:
This should be implemented with a visible 2 part (XY and Z) speedometer, as well as a means to increase or decrease by some amount (or %) the programmed feedrate. (and/or display the differential between actual and programmed values)  

This feature is quite common on commercial CNC machines. 

As I understand, it's usually implemented as a dial that can +/- the programmed (or % of available) feed rate. the available  "-" value should be able to go to zero with a pause (+spindle stop) as the lowest possible value, the + value shouldn't exceed maximum feed rate, and center value should be the programmed feed rate.

GeroBH, 2019-02-15:
> "I'd like to be able to scale the F values on the fly so I can slow down or increase the rate of advance while milling." -Tobor

Do radio buttons or sliders exist in kivy?
So basicly, whatever F is in the g-code is multiplied by a factor 0.1 - 1 (10% - 100%)?
At stage 1 I can accept that it will affect Z, but at stage 2, Z should have it's own slider.
Being able to adjust both on the fly if you feel something is off would give great advantage as compared to stopping and adjusting and reloading the g-code.

mrfugu, 2019-02-15:
I highly recommend taking a long look at some other well used CNC software panels: WinCNC, Mach3, LinuxCNC, as well as the control layouts of various commercial mills (Haas, etc).

They have many common elements that GC is sort of on the same path towards, but could use some more work towards optimizing. 

Similar to the thread on creating more Macro/Function buttons or keys. A concise and highly functional control panel layout is critical to ease of operation...