MaslowCNC/GroundControl #51: G-code Start Location With Gravity

iceblu3710, 2016-11-16:
Have thought of a potential issue that would be an easy fix.

When routing around something if the gcode calls for entry anywhere below the object when the cut starts the object will fall into the bit once complete. This is not an issue with standard machines as gravity plays no role.

The solution would be to have the entry point on all closed objects to start on the coordinate with the lowest Y coordinate.

davidelang, 2016-11-16:
Even on regulat CNC machines you can run into problems with pieces shifting 
after the end of the cut.

And remember that you may be cutting more than one thing, so even if you start 
your cut above one part, it can still fall into the path of another cut you are 
making later in the same sheet for a different part.

The normal fix for this is to design in 'tabs' so that the piece does nto get 
fully detached and then cut through them after the CNC finishes all it's 
cutting.

that will require an active Z axis to do.

Another approach I've seen is to not cut allt he way through the material and 
finish from the other side with a flush cutting bit on a trim router (if you 
leave 1/16 or less of the material, it cuts _really_ fast)

David Lang

BarbourSmith, 2017-01-23:
Ultimately I would like to add CAM to GroundControl so you can open a .dxf or .svg file directly in the GroundControl software and generate G-code. When we can do that I think you are absolutely correct that unless the user specifies otherwise all the cuts should begin at the top.

@davidelang Tabs are the easiest solution 💯

iceblu3710, 2017-01-25:
If you want to add gcode ad dxg/svg loading then take a look at [LaserWeb/LaserWeb4](https://github.com/LaserWeb/LaserWeb4)

I wrote the dxf importer currently in beta. If you talk to the guys there they tried to make everything like the gcode engine as modular as possible.

BarbourSmith, 2017-01-26:
What an awesome project! It looks like LaserWeb packs some pretty serous support for different file types and at the same time the UI looks good. When the time comes to add dxf/svg support, we will absolutely reference how you did it. From some digging around it looks like its possible to call JavaScript functions from Python, but it's ugly. I wish we could just steal your code wholesale. Thanks for pointing that project out, it's a valuable resource.

mrfugu, 2017-08-03:
This issue can be closed.

davidelang, 2017-08-03:
Bar, why don't you grant mrfugu permission on github to close issues.

David Lang

On Thu, 3 Aug 2017, mrfugu wrote:

> Date: Thu, 03 Aug 2017 17:55:32 +0000 (UTC)
> From: mrfugu <notifications@github.com>
> Reply-To: MaslowCNC/GroundControl
>     <reply+0021b24f3c4a15ed7157b9dfce020516f9310791ebac189a92cf00000001159b259
>     492a169ce0b4c56b4@reply.github.com>
> To: MaslowCNC/GroundControl <GroundControl@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Mention <mention@noreply.github.com>
> Subject: Re: [MaslowCNC/GroundControl] G-code Start Location With Gravity
>     (#51)
> 
> This issue can be closed.
>
>

BarbourSmith, 2017-08-03:
Great idea!

@mrfugu are you comfortable with that plan?

closed 2017-08-03 by BarbourSmith