MaslowCNC/GroundControl #59: groundcontrol.kv FrontPage render issue on a Mac with Retina display

blurfl, 2017-01-20:
groundcontrol.kv renders improperly on a Mac with a Retina display, see image below. The same file renders correctly on a non-Retina display.
FrontPage uses some hardcoded widths and heights that seem to be involved.

[image]

BarbourSmith, 2017-01-23:
That's an issue, thanks for catching it.

I've actually run into this one before at a company I used to work for. I think the issue was resolved by switching from defining font sizes in pixels to font sizes in points.

As a quick defense for why I hard coded the dimensions of anything, I did that because when I had the dimensions defined relative to screen size the buttons were too small to click on a small screen like a little tablet. Hard coding them means that the g-code canvas resizes with the window while the buttons stay the same size. I'd really prefer a system where nothing was hard-coded, but I couldn't come up with one.

blurfl, 2017-01-23:
It looks like the latest Kivy addresses this, by adding dp() for padding and spacing and sp() for fonts. I have a version of groundcontrol.kv using dp() that seems to correct this, I can offer that if you like.

davidelang, 2017-01-23:
On Mon, 23 Jan 2017, BarbourSmith wrote:

> That's an issue, thanks for catching it.
>
> I've actually run into this one before at a company I used to work for. I think the issue was resolved by switching from defining font sizes in pixels to [...]

I'll point out that buttons that work on a 3.5" phone screen will be horrible on 
a 10" tablet screen. I understand the problems with making the sizes be based on 
the screen resolution, but I remember seeing that Android has a call to get an 
idea of the physical size of the screen.

But a more generic method would be to have a config option to scale the size, so 
that people can make them the size they want no matter if they are on a 3" or 
100" screen (with a resolution between 720P and 8K), or even have something 
spanning multiple monitors.

David Lang

blurfl, 2017-01-23:
If I understand correctly, that's what dp() and sp() accomplish. I don't have an Android device to test on, but this solves the issue for the various Mac and Windows screens I've tried.

BarbourSmith, 2017-01-23:
@blurfl If you've got a solution that fixes the problem on retina screens, I would love to merge it if you make a PR. I don't have any way to test (not having  a retina screen) so I really appreciate your help.

@davidelang I like the idea of making it adjustable to support the whole 3"-100" range. Kivy offers a built in platform independent calls to get the current screen resolution and current window resolution. If I remember them correctly they are Window.height and Window.width. Maybe the best solution is to use the screen size to make a good guess for how things should look and then let people customize from there. 

There's a lot of things on my todo list right now that are pretty critical so I might not get to that one for a few weeks. Will you make a new issue for it so I don't forget?

blurfl, 2017-01-23:
I'll make a PR shortly. Glad to have a chance to contribute.

blurfl, 2017-01-24:
I’m bumping my nose trying to sync but getting 'access not granted'? I’m new to github, following the example in KS update #13.

BarbourSmith, 2017-01-24:
Hmmm, my first guess is that maybe you are syncing to the main repository instead of to a fork. A fork is your own copy of the code so you shouldn't have an issues with access. That being said, I can see that you are working on a fork here: https://github.com/MaslowCNC/GroundControl/network so it looks like you are doing it right. 

When exactly do you see the message "access not granted"?

blurfl, 2017-01-24:
Got it straightened out. Thanks for your patience.

BarbourSmith, 2017-01-26:
bug found and fixed by @blurfl  in PR #61 . Awesome.

closed 2017-01-26 by BarbourSmith