MaslowCNC/GroundControl #72: Serial connection issue on Mac platform

blurfl, 2017-02-07:
On the Mac platform, GroundControl makes and loses connection to the Mega board. I've verified on two different machines. A windows VM, on the Mac, running Windows python-kivy-GroundControl seems to work correctly. FWIW, a Ubuntu 16.04 VM running on the Mac makes and loses connection in the same manner as the Mac.

blurfl, 2017-02-07:
Using a terminal app instead of python, the connection on the Mac seems to be stable and I'm able to send Gcodes (G20, G21, G0 X0.0 Y0.0, etc.) and get reasonable responses. Is there an issue with pyserial?

blurfl, 2017-02-07:
By the way, I'm using python 2.7.10 on the Mac, 2.7.12 on the Windows VM - does this matter?

blurfl, 2017-02-07:
Updated to python 2.7.12 on the Mac, no joy.

BarbourSmith, 2017-02-09:
I don't have much to contribute other than to say that I can duplicate the issue and will start looking into it. 

I think that the pySerial.readline() function in the file serialPortThread.py line 70 is returning partial lines. I bet this has to do with the way windows and mac terminate lines (off the top of my head I think windows is \n and mac is \r\n but I'm not sure)

Thank you for filing this bug.

BarbourSmith, 2017-02-09:
The issue seems to be caused by the way in which the software tries to connect to the machine automatically. Every two seconds the software tries to connect. If a connection is open on windows an exception is thrown and caught and no new connection is opened. On mac it appears that a new connection is opened, leaving multiple threads trying to read the serial port and infinite sorrow.

blurfl, 2017-02-09:
It sounds like you're on the right track, is there a way I can help?

BarbourSmith, 2017-02-09:
😄 The real question now is what's the cleanest way to fix it. 

The simple way would be to set a flag when the connection is set up and not try to connect unless the connection is lost. I don't really like flags 🏴 . They always seem to end up causing trouble down the road so I'm going to look into seeing if I can poll the connection directly in a thread-safe way. 

I'll put in a pull request once I've got something I feel ok about and I'd love your feedback on how it looks.

BarbourSmith, 2017-02-10:
There is a partial solution ( 🤞 ) to the solution up in #74 . When you have a second I would love your feedback.

blurfl, 2017-02-10:
I’ll give it a try. Funny about the drill motor, as in ha-ha!

blurfl, 2017-02-10:
I can see that the serial port connection process re-iterates when the connection is lost and re-connects properly. Works well on Mac and Win10. 👍

When I capture the serial streams I don't see GroundControl sending to the Mega when I'm running on the Mac (or Win10). Made me wonder if the flag 🏴 'machineIsReadyForData' was preventing the send. To troubleshoot this, I changed the line that sets the initial state of machineIsReadyForData to True and that seems to have made a difference. With that change, on the Mac I can see data scrolling by in the connection panel of kivy and I can change INCHES to MM and back. Funny, after this change, on the Windows machine I only see what the host sends 'G20' or 'G21', but on the Mac I see parts of every line sent by the Mega as well. What's more, it looks like python on the Mac has trouble keeping up with the data flow as several lines go by after I've changed INCHES to MM or back. Finally, It also looks like the Mac version loses connection during the interchange of command and response. If the position data should not show that's one issue. If it should show but is slow to render would be a different one. Should this be a separate Bug report? Which one?

Last, I don't have a drill motor handy to check that part 😀

BarbourSmith, 2017-02-15:
The drill thing turned out to be because I was running the machine, dust collector and drill off of one pretty cheesy extension cord and so turning on the drill was causing the whole machine to power cycle when the voltage got too low. I don't see any effect on the windows laptop for some reason but I bet it has to do with the USB driver chips which is way far down the rabbit hole. The solution is to buy a better extension cord :-).

When the machine is running you should only see lines of g-code, never the positional information. The coordinates should be caught by ground control and displayed in the digital readout and in the cross hair position on the screen. 

I just added a couple more subtle changes to the branch which fixed the issue with reconnecting taking two or three tries. The issue was that the connection timed out before the connection could be fully set up.

I'm in the process of running about an hour cut on Mac right now. About ~45 min in and things are looking good.

Sorry this has been taking my a while, Hannah has the Mac and she "needs it" so I can only steal it intermittently.

blurfl, 2017-02-15:
I understand about sharing a computer - I’ll take a look at the branch when I get the Mac back, it’s running my wife’s kiln tonight. I wish I knew for sure that the R.Pi touchscreen would work with GroundControl.

BarbourSmith, 2017-02-15:
That seems like a very honorable thing for a computer to be doing.

Which model of R.Pi touch screen do you want to use? I'll order one and make sure it works. I need to get one to test anyway, it might as well be the model you want to use.

blurfl, 2017-02-15:
Not sure how important screen size will be. The small ones have very limited resolution, and some Linux things won't limit themselves to the small screen. Wind up with much of the window unreachable😕. They are much cheaper than the official RPi touchscreen though.
 I'll probably just go with the official one, though. I'd rather focus on the CNC part to begin with.

BarbourSmith, 2017-02-15:
I've got the official Raspberry Pi touch screen (7") on the way so I should have it by next week. I'll keep you in the loop with how it goes.

blurfl, 2017-02-16:
This seems to have solved the Mac connectivity issue. I think you're done, should I close the issue?

BarbourSmith, 2017-02-16:
Fantastic! I'll merge the pull request. 👍 

After you close this issue, feel free to make a new issue for testing with the raspberry pi touch screen and I'll keep you in the loop as I get that set up.

blurfl, 2017-02-16:
Right-I, will do.

closed 2017-02-16 by blurfl