MaslowCNC/GroundControl #76: Test GroundControl with Raspberry Pi touchscreen

blurfl, 2017-02-16:
The Raspberry Pi platform is capable of running Kivy, and can be configured with a touch screen. This configuration is tempting for mounting on the machine sled itself.
 Earlier attempts to run GroundControl have run into various issues including cursor visibility. 
 Assemble a test unit and determine what issues need to be addressed in order to run the machine successfully, whether it is reasonable to mount the Pi on the sled and what compromises this configuration brings.

jbnimble, 2017-02-16:
I am interested in seeing what issues arise with sled mounting of additional hardware. I have been thinking of ways to mount all the electronics (tablet, arduino, etc) including the chain motors to the sled. Having 1 power cable to a power strip and the vacuum hose be on the sled seems like it would have a lot of advantages from an accessibility and emergency stop perspective.

davidelang, 2017-02-16:
On Thu, 16 Feb 2017, Jesse wrote:

> I am interested in seeing what issues arise with sled mounting of additional
> hardware. I have been thinking of ways to mount all the electronics (tablet,
> arduino, etc) including the chain motors [...]

you trade of the minor complexity of one power cord (that can just drag along 
the floor) for the need to have multiple wires going up to each motor.

how are you going to keep these wires out of the way?

David Lang

blurfl, 2017-02-16:
You've got a good point about cable management. The wires to the motors/encoders don't have to follow the path of the chains, of course. Looking at Kickstarter update #10 is seems @Bar has given this cabling a lot of thought, they don't look too unmanageable.

jbnimble, 2017-02-16:
My initial thought was to have everything on the sled, but maybe that is not realistic.

davidelang, 2017-02-16:
On Thu, 16 Feb 2017, Jesse wrote:

> My initial thought was to have everything on the sled, but maybe that is not realistic.

you really don't want to be trying to hit a button or read something mounted to 
a sled that's moving around.

blurfl, 2017-02-16:
Not a problem at slower feed rates if the GUI makes the 'PAUSE' button large enough :) If one could VNC into the Pi, that could help. Kivy over VNC seems to be a different issue, though.
Having one object to stow when putting the machine away would be nice. Some of the measurement/configuration routines that @Bar has been looking at might need a different approach, though. Many things to try, much to learn.

bar, 2017-02-17:
lovely, people, what are you talking about? 

I believe, you might be talking about a different person

blurfl, 2017-02-17:
That’s the case - the thread should be including the project leader, @BarbourSmith but I've used the wrong nick for him so you’ve gotten included instead.
 I'm sorry this happened and for the bother it has caused.

BarbourSmith, 2017-02-28:
I've got the software running with touch support and mouse support working (after much fiddling). I will make a copy of my disk image so others can download it. I know @blurfl has his working too (I think), but it may be of use to others.

blurfl, 2017-02-28:
I did get it to run, with touch and with the mouse though the circle cursor shows at the location. The image pane seemed to be larger than the screen at the bottom and the code text window seemed to extend past the bottom of the screen. I used the kivypie image but had to switch it to python2.
I'll look forward to trying your image.

BarbourSmith, 2017-02-28:
😃 Mine is kivypie with python two added so it's probably exactly the same :-). I added a little script so the pie boots into GC so you don't have to log in but other than that it's pretty stock. 

I have the same UI resizing awkwardness that you saw and it's even worse with the Android version running on my phone so I'll have to tweak the way things resize.

blurfl, 2017-02-28:
I did the 'boot into GC' thing too, but found I couldn't ctrl-C out to the command line. Might have been my keyboard. Reminds me that a 'Quit' button might be useful among the GC controls...

BarbourSmith, 2017-02-28:
I think I read in one of the forums that there is a bug in Kivy that prevents you from being able to control-c quit. 

I will file a bug to add a "Quit" button.

As a work around I found that there is a bug that pressing escape crashes GC so you can get out that way (if you have a keyboard attached).

blurfl, 2017-02-28:
So, an 'escape' plan... 😛

blurfl, 2017-03-01:
I notice that I get double input in the speed input frame - '3' becomes '33', backspace deletes both characters.

BarbourSmith, 2017-04-01:
I know there are other issues described in this thread, but I think they each have their own issues filed. I believe the original issue of testing GC on the R Pi touch screen is good to go at this point. Anyone disagree?

closed 2017-04-01 by BarbourSmith