MaslowCNC/GroundControl #78: build-OSX branch groundcontrol.app package fails to run

blurfl, 2017-02-23:
It looks like the script 'Resources/script' is looking for yourapp/main.py instead of cncgc.py. I don't understand much about buildozer; I found that duplicating cncgcx.py and renaming it main.py allows the package to run.

jbnimble, 2017-02-23:
I described the steps I had to take to get it to work on Android here http://www.maslowcnc.com/forums/#!/software-issues:problems-running-ground-con which included having to change it to main.py

blurfl, 2017-02-24:
Buildozer seems to require a file 'main.py' - from the specifications:
"The location must be a directory that contains a main.py file."

Unless there is a reason, perhaps we should rename cncgc.py as main.py in the source?

BarbourSmith, 2017-02-24:
I think that is a more standard and more clear system anyway even if buildozer didn't require it. I'm renaming the file right now.

@blurfl Were you able to get it compile into a single package after renaming the file? That's the first thing on my todo list for today.

@jbnimble That's amazing that you got GC to run on Android. Awesome job!

blurfl, 2017-02-24:
Yes, it compiles and runs nicely. Couldn't get a Maslow icon built into it though😄

BarbourSmith, 2017-02-24:
Beautiful! 🤞 I can replicate your success today. 😃

BarbourSmith, 2017-02-24:
@blurfl Did you have to install anything special to get it to build? I keep getting a cryptic CalledProcessError: returned non-zero exit status 1 issue but I can't for the life of me figure out what I'm missing

blurfl, 2017-02-24:
We’re talking about the build-OSX branch, yes?
 
First I ran it in terminal using ‘kivy main.py’ from within the appropriate GroundControl directory, to ‘pre-flight’ itn- all OK. I know almost nothing about how bulldozer works, so here are the steps I took next:

> bulldozer osx clean
> bulldozer osx update
> bulldozer osx release
> bulldozer osx deploy
> open bin/groundcontrol.dmg
Copied into /Applications and it ran…

closed 2017-03-07 by blurfl

BarbourSmith, 2017-03-15:
@blurfl I haven't had a chance to sit down with my mac compiler and figure out why it's unhappy, would you be willing to share your compiled version of GC?

blurfl, 2017-03-15:
Glad to. I see you merged it recently, I updated and built the .dmg just now. Tested OK on a Mac with no Kivy, virus checked with ClamAV. How shall I get it to you (50mb)? No Dropbox here...

BarbourSmith, 2017-03-15:
You are a life saver. Do you use gmail? Any chance you can email it to me bar@maslowcnc.com?

BarbourSmith, 2017-03-15:
Got it. Thanks. 👍

blurfl, 2017-03-15:
Which method worked?

BarbourSmith, 2017-03-15:
The shared google folder. I think the iCloud would have worked if I were on Mac, but for some reason it wouldn't let me download the file on windows

blurfl, 2017-03-15:
Good to know, thanks.