MaslowCNC/GroundControl #79: openFile defaults to a directory within the groundcontrol app

blurfl, 2017-02-24:
When running for the first time, or without a gcode file selected, or when the gcode file cannot be found, the 'openFile’ dialog starts within the file structure of groundcontrol itself. This can be confusing.
 Instead, use the user’s home directory as the starting point. A two-line change to viewMenu would set the path to path.expanduser('~') if the gcode file is not found. 
 I've tested is on OS X and Windows10, but not Android.

BarbourSmith, 2017-02-24:
I think this is a great idea. It's especially important not to confuse people the first time they open the program. If you make a pull request for the change, I would love to merge it.

blurfl, 2017-02-24:
I was starting to look at the FileBrowser widget as well. I'll do a PR for just this change, and look into that later.

blurfl, 2017-03-08:
The issue is present once again, but I'm not certain at what point it re-emerged.

BarbourSmith, 2017-03-09:
Hmmm. It may be related to the .ini file issue. Having a .ini file would mask the problem by providing a more meaningful default path.

blurfl, 2017-03-09:
I've gat a fix, can do a or later today...

closed 2017-03-14 by blurfl