MaslowCNC/GroundControl #94: groundcontrol.ini missing

blurfl, 2017-03-08:
The fork comes without a groundcontrol.ini, this causes crashes:

The first run crashes immediately but creates an .ini file. This file has 'False' as the value for zaxis and has no value for openfile. The crash is related to:

UIElements/frontPage.py", line 131, in moveGcodeIndex
     gCodeLine = self.data.gcode[self.data.gcodeIndex]
 IndexError: list index out of range

The second run opens successfully and many parts work as expected, but it crashes when a gcode file is chosen and 'load' is clicked. The filename chosen is not written to the .ini file and the crash shows the same error.

 The file with full path can be entered in the Settings panel, or at /Applications/groundcontrol.app/Contents/Resources/yourapp/groundcontrol.ini with a text editor.

blurfl, 2017-03-09:
Seems to come from "Pr 92 - rev 2 #93". I think that frontPage.py doesn't have access to the gcode list built by viewMenu. My python chops are inadequate, but I'll keep digging :)

blurfl, 2017-03-09:
@BarbourSmith, I've kludged up a fix by copying the code from within viewmenu.reloadGcode into the beginning of frontpage.moveGcodeIndex. This creates a local version of the gcode list. Not sure if that is a fix or a bandaid though. What to do? Do you want to see this, or do you have a different idea for fixing the issue?

BarbourSmith, 2017-03-09:
Thanks for catching this. 

I'm going to dig into it right now. Sorry for the slow response, I've been writing up assembly instructions! 😃

BarbourSmith, 2017-03-09:
The .gitignore file removes the .ini file from commits automatically but it remains on our respective computers so I believe this issue applies to all branches when they are first downloaded.

It looks like what's going on is that I changed so that when a new file is loaded it resets the indicator to the beginning of the file, but if there is no .ini file, there is no gcode to load and so it crashes when I tries to go to the beginning because there is no beginning.

blurfl, 2017-03-09:
I've got a fix, can do a or later today...

closed 2017-03-10 by blurfl