MaslowCNC/GroundControl #114: Actions tab breaks serial connection

blurfl, 2017-03-20:
Upon activating the Actions tab, a call to self.connectmenu.updatePorts() found in UIElements/otherFeatures.py breaks the serial connection, at least under OSX. Commenting out the offending line is a bandaid.
 I've tried the following patch with success, avoiding the use of self.data.serialPort:
'

    def updatePorts(self, *args):
        
        portsList = ["Available Ports:"]
        
        if sys.platform.startswith('win'):
        	sysports = ['COM%s' % (i + 1) for i in range(256)]
        elif sys.platform.startswith('linux') or sys.platform.startswith('cygwin'):	        # this excludes your current terminal "/dev/tty"
        	sysports = glob.glob('/dev/tty[A-Za-z]*')
        elif sys.platform.startswith('darwin'):
        	sysports = glob.glob('/dev/tty.*')
        else:
        	raise EnvironmentError('Unsupported platform')	        
        # for port in self.data.serialPort.listSerialPorts():
        for port in sysports:
        	portsList.append(port)
        
        if len(portsList) == 1:
            portsList.append("None")
        
        self.COMports = portsList

    '

This should probably be a patch in the main branch, but I haven't had time to test it on other platforms yet.

BarbourSmith, 2017-03-20:
🔥 🔥 🔥 👍 👍 👍 

I'd say make a PR against master. I can check it out on windows at least. I've been imagining the Build-OSX branch as only being different from master in terms of things particular to the build process.

blurfl, 2017-03-20:
Right-o, will do.

BarbourSmith, 2017-03-22:
Have we confirmed that the issue is solved?

blurfl, 2017-03-22:
I say 'yes'

closed 2017-03-22 by BarbourSmith