MaslowCNC/GroundControl #126: allow machine dimensions to be entered in metric or imperial

davidelang, 2017-03-25:
add a units button to ground control and let someone enter the measurements in whatever units they are measured in and let the software do the conversion

BarbourSmith, 2017-03-25:
This is with regard to entering the machine dimensions, correct?

davidelang, 2017-03-25:
Yes, when you enter the mchine dimensions into Ground Control, it should have a 
metric/imperial toggle and do the conversion between the two internally.

David Lang

BarbourSmith, 2017-05-13:
@davidelang 

Since the new calibration accepts measurements in either mm or inches, are you comfortable with closing this?

davidelang, 2017-05-14:
yes.

On Sat, 13 May 2017, BarbourSmith wrote:

> Date: Sat, 13 May 2017 16:20:57 -0700
> From: BarbourSmith <notifications@github.com>
> Reply-To: MaslowCNC/GroundControl
>     <reply+0021b24f0cf4c2f274f39741119efc22891b0b862638f3d892cf00000001152f56d
>     992a169ce0cee5b1b@reply.github.com>
> To: MaslowCNC/GroundControl <GroundControl@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Mention <mention@noreply.github.com>
> Subject: Re: [MaslowCNC/GroundControl] allow machine dimensions to be entered
>     in metric or imperial (#126)
> 
> @davidelang
>
> Since the new calibration accepts measurements in either mm or inches, are you comfortable with closing this?
>
>

closed 2017-05-14 by BarbourSmith