MaslowCNC/GroundControl #127: create a machine self measurement routine

davidelang, 2017-03-25:
At one point @BarbourSmith described a machine calibration routine that measured the distance between the motors, it would go something like

1. position both motors with a tooth on the gear pointing straight up.

2. tell ground control how many teeth on the gears

3. hook a chain across the top of one gear with the slack to the outside.

4. have the machine run the motor to take up the slack (feeding the chain  out towards the center of the machine). This could either be a continuous feed as long as a button is held down, or a series of 'feed 1 foot of chain' presses.

5. when enough chain has been let out to hook it over the top of the second motor, hook it there and tell ground control that it's in place

6. ground control would then rotate both motors (feeding more chain from the first, taking up chain from the second, until you have a good wrap around the second motor, say 6=12" worth to be safe)

7. ground control would then instruct both motors to start pulling against each other (adding tension to the chain), until there is a good amount of tension in place.

by knowing how much chain was let out by the first motor, and then how much was pulled back when tensioning the chain, you should know exactly how much chain there is between the motors. And since each link is .25", the exact distance between the motors can be calculated.

Bonus items:

if someone is willing to count links, steps 3-6 could be replaced by "tell me how many links there are between the two that you used to go from the top of one gear to the top of the other

the extra pins on the arduino could be used for a switch to feed out the motors for step 4 instead of having to hit something on the controlling computer.

davidelang, 2017-03-25:
duplicate of #73 (mostly)

BarbourSmith, 2017-05-13:
Added

closed 2017-05-13 by BarbourSmith