MaslowCNC/GroundControl #130: Resolve carriage stop after "J" in gcode

nwrancher, 2017-03-25:
Current configuration of Ground Control will pause after J commands in the g code.  I was able to replicate the behavior using both the Dragon.nc and BiggerMotorMount25in.nc files found in the "gcodefortesting" folder in GroundControl-Master.

johnbenweber, 2017-03-27:
Can you describe the behavior? I'm just trying to get an idea of whether the issue is being caused by GroundControl or the Firmware. When you say pause do you mean that the machine briefly stops between commands but resumes on its own or does the execution of the gcode file pause requiring you to press the "run" button or something else?

nwrancher, 2017-03-27:
The execution of the file pauses and requires pause/run to continue.  If 
time has passed it continues with loss of position.

BarbourSmith, 2017-04-01:
Is this issue still open? I haven't heard about it in a little while, but it's a top priority bug if it's still active and I don't want to neglect it.

nwrancher, 2017-04-01:
Resolved with changing to GRBL gcode output.

closed 2017-04-01 by nwrancher