MaslowCNC/GroundControl #131: Branch '-switch-to-using-dist-between-motors' vertical control error

blurfl, 2017-03-25:
Downloaded .zip of the branch and latest firmware. Ran 'cal. chain lengths' and input the values in the settings panel. Noticed that Action/Return-to-Center was low. Up-arrow sends sled down. Down-arrow sends sled down. It seems like all arrows always pay chain out on both motors.

BarbourSmith, 2017-03-25:
Very interesting. Good catch before we merged it. 

I'll see if I can replicate the same behavior.

blurfl, 2017-03-25:
I got pulled away before I could diff the folders, sorry. I hope it's just me :)

BarbourSmith, 2017-03-25:
Looks good to me. I one possibility is that if somehow you were to run one of the new branches without the other, it would give you the kind of strange behavior you saw. I know you know this already because you stated in the issue that you updated both.

I was holding off merging them because they have to happen at the same time and I was worried we would cause lots of confusion as people updated one, but not the other and got weird results. I think that the solution is to just not use the same letter to send the length of the new distBetweenMotors as we used before to send the motorXOffset. If we use a new letter, the whole system can be backwards compatible and I can merge it right now :-)

blurfl, 2017-03-25:
Sounds good, merge away!

blurfl, 2017-03-26:
I think the .ini file was the culprit. Something to think about if we see .ini files between releases.

BarbourSmith, 2017-03-26:
Good thinking!

blurfl, 2017-03-26:
The issue was due to a version mismatch between GroundControl and the .ini file.

closed 2017-03-26 by blurfl