MaslowCNC/GroundControl #134: Strange jogging behavior after failed Gcode

chadheiser, 2017-03-26:
[Gcode1.zip](https://github.com/MaslowCNC/GroundControl/files/870549/Gcode1.zip)

When I run MakerCam generated gcode for the Arm_Front.svg (both in attached zip) I have an issue where the sled moves beyond the 4'x8' workspace while Ground Control shows 18"+ of clearance. Upon stopping the cut the jog commands ignore or multiply the distance setting. If I set it to jog 1 inch down, it will move down and simply continue until I hit stop (16"+ and no signs of stopping.) I have to restart Ground Control to reset the job commands.

I have followed the same steps with the attached BrickHolder.nc file, and that cuts as expected and has no effect on jogging before, during or after.

BarbourSmith, 2017-03-26:
possibly related to #130 ?

chadheiser, 2017-03-26:
Was able to get the part to cut if I rotate the SVG in MakerCAM and generate Gcode. Attached is the .nc that works.
[armFront_rotated1.nc.txt](https://github.com/MaslowCNC/GroundControl/files/870968/armFront_rotated1.nc.txt)

BarbourSmith, 2017-03-26:
I'm having trouble replicating the issue on my machine which I guess is a good thing in the long run, but it makes it hard to figure out what's going on. Where on the board was the goal to cut the part from? 
[image]

chadheiser, 2017-03-26:
We tried cutting in the bottom left corner. Due to our sled having the two bricks and a vaccum hose we were about 8" from the bottom and maybe 12" from the left. As soon as the sled moved towards the hole it would take off and leave the cutting area quickly, after stopping it is when we would have trouble with the jog commands.

johnbenweber, 2017-03-27:
Can you identify the last gcode line that was run before the sled took off?

BarbourSmith, 2017-03-27:
I asked about that in the original forum post and the line was something like G0 X11 Y25 or something similar so the machine was doing what it was told, for some reason Ground Control was telling it to go to the wrong place...at least that's my working hypothesis

chadheiser, 2017-03-27:
Sorry, I saw this early this am and forgot to update once I got to work.

Line 10 (the first move command)
G0 X22.0273 Y27.8897

It looked like it was headed in the right direction, but it was like the positioning was off since it moved much farther than GC showed it moving.

I haven't really had a chance to debug this too much more other than rotating the SVG and re-exporting. I have now updated to the latest compiled Windows copy of GC and I've updated the firmware to latest (as of yesterday). We're going to assemble the legs today and try to get the frame closer to assembled, but if I get a chance I'll run this file against the latest GC/Firmware. Will update.

nwrancher, 2017-03-27:
I just ran latest GC/Firmware and on mine it exhibited the same behavior.  Home was way off and it kept trying to tear the sled apart.   Jogging was always up trying to find position.  Files paused and lost position.

BarbourSmith, 2017-03-27:
I'll run the same on mine and see if I can do it too

nwrancher, 2017-03-27:
Dragon.nc file was my test.

BarbourSmith, 2017-03-27:
I'm still having a hard time making it happen. Walk me through exactly what you do to set it up. What I did was:

- Open dragon.nc
- Move the machine down and left 12 inches
- Redefine the origin there
- Click "Run"

My dragon then seems to cut normally. What am I doing wrong?

nwrancher, 2017-03-27:
I don't know,  I didn't redefine the origin, but I can't get 4% into it before I stall and position starts slipping.

BarbourSmith, 2017-03-27:
So when you click "run" it gets to 4% and then the machine stops moving, but the position indicator continues?

BarbourSmith, 2017-03-27:
4% looks like it's right around where the first spike on the back is

nwrancher, 2017-03-27:
I think it was 3.3%, after a J command for sure.  It all just stops and does nothing.  If I hit Pause/Run quickly it will resume.  If I wait, then Pause/Run, the position indicator will move while the sled does not.

I didn't make it to the spike, not even the bottom tab.  First stall is right before the tab, next was after the corner before the spikes, then again somewhere on the first spike, then I quit.

BarbourSmith, 2017-03-27:
I'm stumped. This is a weird one.

The only thing I can think that is different between your machine and mine right now is the settings.

Can you screenshot tell me what your settings are so I can set mine up exactly the same?

nwrancher, 2017-03-27:
Settings were all over the place for me again today, because new GC went back to default and I was left guessing at how many mm 10' is again.  I think I used the defaults, but set vertical to anywhere from 270-380.  No matter what I did, it thought home was above the sheet.  I ended up jumping the chains out of frustration and thinking I was going to destroy all of it.  So, I set home by slipping the chains manually.  Settings weren't helping today with the new firmware.  It was better the other day, but the file behavior was identical.  Sorry man.  Lots of fails for me today and I don't know where to start troubleshooting.  I'm definitely going backwards with each iteration though.

BarbourSmith, 2017-03-27:
That's not good, we want forward.

I think the core issue here is something with the settings. If it thinks home is above the machine it's going to do not what we want. What behavior do you see with these settings: 
[image]

nwrancher, 2017-03-27:
If I'm not mistaken those are the defaults.  It tried to climb off the sheet.

The other day I could change home with settings.  Today it did not seem to respond to setting changes.  I tried extreme settings and home stayed way up there.

nwrancher, 2017-03-27:
After that happened on the defaults I re-measured the chains, re-calibrated the motors, and it ended pretty much at home.  Until I pressed Home.  Then it headed North.

BarbourSmith, 2017-03-27:
That sounds like exactly the right thing to do.

Just for the sake of sanity let's do this very step by step and see where your machine does something different than mine.

First, let's make sure we're running the same version of everything. Are you running GC from the source version or the pre-compiled version? Let's both update our firmware to the very latest from this morning.

nwrancher, 2017-03-27:
I am using the precompiled version from a day or two ago.

johnbenweber, 2017-03-27:
I ran into a similar issue when I was setting my hardware up. Basically I had one of the encoders hooked up backward so the steps were in the opposite direction of that expected by the pid controller. For some reason this caused both motors to go full bore upwards no matter what.

johnbenweber, 2017-03-27:
I'm not sure if that's possible with the official kit but have you double checked all of your wiring? Is it possible one of your encoders has failed?

nwrancher, 2017-03-27:
I don't know what an encoder is?

nwrancher, 2017-03-27:
But.....I haven't messed with the wiring since day one, and even now, it does some stuff fine.

BarbourSmith, 2017-03-27:
Interesting. @nwrancher shouldn't be able to have that problem because the encoders are built into the motors, but it's a good thing to keep in mind.

The encoders are the part of the motor assembly which keeps track of how much they have moved.

@nwrancher what do you see if you run the "Test Motors" action under the "Actions" menu?

I'm going to compile you the very latest version right now so that we can test with the very latest code.

nwrancher, 2017-03-27:
I have the laptop with GC here with me, but the machine and Arduino are down in my shop, so I can't test motors at the moment.  I was worried they were damaged when it was pulling the sled up so high, but I calibrated after that and they passed just fine.  I will go down there a bit later, I just got the email that the Z is in the mailbox!  Let me know what tests, and I'll bring the new firmware and what not.

Thank you for the new version.

johnbenweber, 2017-03-27:
@BarbourSmith  I understand, just throwing that out there since it I don't think I ever documented that problem.

BarbourSmith, 2017-03-27:
@johnbenweber I had that same problem early on too. I think the "test motors" will help diagnose the issue, but it's still a tricky one to figure out what's going on.

@nwrancher 
Sounds perfect.

Good to know that the motors pass the test motors test, that means all the electronics are ✔️ 

Try this version [here](https://github.com/MaslowCNC/GroundControl/blob/build/GroundControl-Windows%20Portable-NWRancher.zip) with all the default settings. I'm going to go through the process of setting it all up and post each step so when you get to the machine you can do the same steps and see where we get different results

BarbourSmith, 2017-03-27:
1. Install latest firmware from https://github.com/MaslowCNC/Firmware
2. Open GC
3. Connect to the right port
4. Check that "Firmware version 0.61" prints on the text part of GC (thanks @johnbenweber )
5. Under 'Actions' click 'About' and see that the version is 0.63
6. Hook the first link of each chain onto the top of each sprocket and click "Actions -> Calibrate Chain Lengths" 
7. See "measuring out left chain" on terminal in Ground Control and left motor spins slowly
8. See "measuring out right chain" on terminal in Ground Control and right motor spins slowly
9. Hook up sled
10. Click "Home" and sled goes relatively close to center (because we're using the default settings it won't be perfect
11. Clicking "Home" repeatedly has no effect
12. Open the "dragon.nc" file
13. Click "Run"
14. Click "continue" on popup about raising bit
15. Machine moves to start cut
16. Click continue on popup about lowering bit
17. Machine follows path

Let me know where your machine behaves differently and we can narrow in on what's wrong

nwrancher, 2017-03-27:
That's helpful Bar, I'll do it.  Hopefully this afternoon.  

I've never seen popups for raising or lowering.  That's why I was confused about the holes in the sled.  Are there supposed to be notifications?

BarbourSmith, 2017-03-27:
You might not see popups on some older versions of the firmware, but on the latest you should be seeing them. If you don't see them at the times indicated in the list, let me know and we will investigate what's happening.

nwrancher, 2017-03-28:
Well, no luck Bar.  I tried according to your instructions, and firmware was .60 not .61.  I could not change that despite my best efforts with the latest firmware master.  

Measured chain length, attached sled, hit home, it tried to tear itself apart again.

BarbourSmith, 2017-04-01:
Are we still seeing this behavior? I was never able to replicate it, so I haven't found the problem yet, but I don't want to neglect it if peoples machines are still going rogue

nwrancher, 2017-04-01:
I believe that for me this was a function of both old firmware and 
losing position from file handling with the HAAS gcode.  All is well now 
for me.

BarbourSmith, 2017-04-01:
Fantastic!

closed 2017-04-01 by BarbourSmith