MaslowCNC/GroundControl #138: Create a version which can be displayed by clicking the 'About' button

BarbourSmith, 2017-03-26:
The "About" button under diagnostics is currently disabled. Clicking it should open a popup which will have information about the version number.

blurfl, 2017-03-26:
How about having the firmware send its version number as well and show both since they are inter-tied?

johnbenweber, 2017-03-27:
See pull request #141. @blurfl it would be good to also show the firmware version. The tricky part is that you have to send a command and wait for a response to get that. My idea was to just print the firmware version as part of the connection process just before it changes units (see #142).

blurfl, 2017-03-27:
How about sending the B05 from the 'About' pop up, then the info will be visible when the pop up is cleared?

BarbourSmith, 2017-03-27:
I agree with @blurfl that it would be ideal to be able to check your firmware version from the "About" popup, but I also agree with @johnbenweber  that it's a little bit complicated to do that right.

I think the way it's setup now where it prints when you first connect to the board is great. It's simple. We might down the road want to add a check in Ground Control to insist that the firmware version is x or greater or something, but for now I like this system.

closed 2017-03-27 by BarbourSmith

blurfl, 2017-03-27:
On the Rasp.Pi with the ‘official’ touch screen, the most recent lines of the Machine Feedback pane are off screen.  I opened #144 on this.
 This would put the firmware version number out of sight.