MaslowCNC/GroundControl #139: Problem installing firmware

ledsled71, 2017-03-26:
Got the Arduino installed, now on step 3, installing firmware.  Got it downloaded, but when I double-click the cnc file, it opens Arduino, but when I click upload, it gives me an error Arduino: 1.8.2 (Windows Store 1.8.2.0) (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\LaPrade\AppData\Local\Temp\Temp1_Firmware-master (2).zip\Firmware-master\cnc_ctrl_v1\cnc_ctrl_v1.ino:16:27: fatal error: CNC_Functions.h: No such file or directory

 #include "CNC_Functions.h"

                           ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
 '

ledsled71, 2017-03-26:
Figured it out.  Might want to put a note in the instructions to extract the files from the zip

closed 2017-03-26 by ledsled71

BarbourSmith, 2017-03-27:
I will update instructions, thank you for pointing that out.