MaslowCNC/GroundControl #144: Machine Feedback window size on Raspberry Pi

blurfl, 2017-03-27:
The most recent few lines of text are below the viewing area when using the 'official' touch screen.

BarbourSmith, 2017-03-27:
+1 I have this also

BarbourSmith, 2017-03-27:
I set things up so that the text/buttons wouldn't scale with the screen size to keep them big and easy to press. I'm not sure that was a good idea, maybe if they scaled with the window they would still be big enough to click.

blurfl, 2017-03-27:
Wasn't going to say anything, thought it might be my setup. Big buttons are nice, but seeing all the machine info is important 😕

johnbenweber, 2017-03-27:
If you are pressed for space we could maybe detect that and switch to a tabbed layout?

BarbourSmith, 2017-03-27:
Yeah, I like the tabbed layout idea. I'd like to do an Android app soon which will almost always run on small screens so it's going to be important. 

I think this is going to be one of those things that we're going to have to use for a while before we really get a sense of what is important to see and when

BarbourSmith, 2017-03-30:
#159 is a duplicate of this so I am going to close that issue:

I see a couple ways to resolve this. Either we make the buttons smaller, or we split the UI into multiple tabs or viewing areas.

I think this is an important issue because I'd like to make an Android App soon too and we're going to see all of these issues there also.

Kivy will let us scale buttons (and everything else) in one of two ways, it can either be scaled as a percentage of the screen size, or set to an absolute size. I went with absolute size because as a percentage of the screen the buttons looked too big on big screens and too small on little screens, but that means that if the screen size is less than the absolute size of the buttons, they get cut off.

If we decide to split things into tabs, I think the big question is what goes on in each view and how many are there. We can do separate viewing areas using either a spinner widget (built into kivy, works well) or something a little slicker like [this](https://www.youtube.com/watch?v=0nlUIbQrf5k). Kivy can give us the [window size](https://groups.google.com/forum/#!topic/kivy-users/BNmZxiWd5Mc) which could be used to switch automatically between little mode and big mode.

What does everyone think? What's the best way to go about it?

davidelang, 2017-03-30:
I think we are going to end up needing tabs at some point anyway, so plan for 
that.