MaslowCNC/GroundControl #149: Ground Control interpretation errors with G-code generated using GRBL Generic format

nwrancher, 2017-03-27:
.NC files generated with GRBL Generic post processing have open paths and errors.

I don't know how to attach a .nc file?  Format not supported for Github upload.  I'll email you.

BarbourSmith, 2017-03-27:
It's really annoying that we can't attach gcode files, here it is in a .zip file
[GRBL generic.zip](https://github.com/MaslowCNC/GroundControl/files/874074/GRBL.generic.zip)

BarbourSmith, 2017-03-27:
I've seen this issue before and I started working on it in firmware PR #120 , needs to be tested against the firmware to make sure the new commands can be read on the Arduino side.

BarbourSmith, 2017-03-28:
This is actually a completely seperate issue than #120 , the problem here has to do with the gcode generator dropping I, J or even X or Y values if they don't change from the previous line.

BarbourSmith, 2017-03-28:
On the ground control side this is fixed in PR #151 , When running the code it cuts almost correctly, but skips a few arcs. Will investigate on the firmware side.

BarbourSmith, 2017-03-28:
Firmware pull request [#155](https://github.com/MaslowCNC/Firmware/pull/155) adds similar support on the firmware side.

G18 commands (switch from XY to YZ plane) are still not supported and might give you a little bit of strange behavior, but it seems like the CAM processor just throws them in there when raising and lowering the bit and they don't seem to do much so we can probably get away with ignoring them for now. We'll see what Autodesk says when they get back to us about keeping G18 out of the code.

@nwrancher , When you get a chance will you let us know if the latest firmware and Ground Control works for you too?

nwrancher, 2017-03-28:
Yes, I'm just up from the shop.  All is well with GC and the Firmware, and your example files.  Still no luck with any of the Fusion360 generated files.  The closest ones are the HAAS-generic, but still there are errors in code handling.  Path looks fine in GC, but tool does not follow path line in all details.

BarbourSmith, 2017-03-28:
Woo! This feels like forward at least.

BarbourSmith, 2017-03-28:
Have you updated to the very latest firmware? I only fixed the grbl-generic file around 3pm so it might be worth updating again

nwrancher, 2017-03-28:
No, not yet.  

Did you install Fusion360 on your computer, or one you have access to where you are?  I wonder if I might include you on my project, and you could do post process attempts and see how GC handles the changes.  In other words, you'd have access to a finished design and could just generate gcode locally using the various options.  Fusion has all the designs online, so I would just include your email in the project.  I think.

Does that seem like it has any value?  I thought it might offer you some insights into the best way to proceed, but....I dunno.

nwrancher, 2017-03-29:
Bar, I just noticed the "0 offset for circles" change from yesterday.  Now the GRBL-generic file that was all error circles yesterday in the GC view, looks like it will cut!  I'll go try and report back.

BarbourSmith, 2017-03-29:
Fantastic! Forward!

BarbourSmith, 2017-03-30:
Are we comfortable closing this?

nwrancher, 2017-03-30:
I think so Bar.  I'm just up from the shop, things seem to be going smoothly with the file handling.  The Z stuff is for sure part of the set up, I was able to modify it today.  I think this is a done deal.  Thank you!

BarbourSmith, 2017-03-30:
Wooooo! Keep us posted if you see an strange behavior again. Thank you again for filing the issue for this. It was a good one.

closed 2017-03-30 by BarbourSmith