MaslowCNC/GroundControl #150: override chain lengths

jessbussert, 2017-03-27:
We need a way to override the computed chain lengths for machines with unusual geometries.

davidelang, 2017-03-27:
can you give an example of a setup that gets this wrong?

On Mon, 27 Mar 2017, jessbussert wrote:

> We need a way to override the computed chain lengths for machines with unusual geometries.
>
>
>

jessbussert, 2017-03-27:
Hi David,
This is more of a placeholder issue than something I need other users to mess with. My machine is nonstandard being that my motors are attached to my sled instead of the frame.  As such, the chain length calibration routine doesn't work for me. Bar and I spoke about this today and he is going to add a special gcode command to let me override the chain length.  He wanted me to post an issue as a reminder.

davidelang, 2017-03-27:
Ok, it seems to me that a better fix is to figure out a calibration routine (i.e. #127 ) rather than having people need to enter things manually. This is a good temporary work-around, but such things have a way of being permanent :-)

jessbussert, 2017-03-27:
Yeah, I agree.  That said, since mine is the red-headed step child of Maslow, I don't want to make anyone spend too much time on workarounds that only benefit me.  If my setup works as I hope it will, then perhaps it will eventually become a supported mod.  In short what I have done is to mount my motors on large bearing rings centered on the router bit.  This way my force vectors remain a simple triangle.  My hope is that I'll be able to garner greater speed and accuracy with this design.  I'll post photos once I get it cutting.

BarbourSmith, 2017-03-27:
It's still a community issue, we're all working on this together so it's important that everyone understand why we're adding the features we are. I agree with @davidelang that it's important not to add things carelessly, because they will be around forever. I've tried to add this in a way that I think is clear. I don't think we should modify Ground Control to allow easy access to this feature because it will confuse everyone and cause a great deal of headache.

The general idea (as I understand it) is that the chains make up only a part of the total length of what suspends the sled in @jessbussert design, there is also a section of rigid arm which can swing, but can't pass over the sprocket. It will be tough to make the auto calibration routine ( #127 ) work for this system.

I've added the command `B06` in firmware PR [#152](https://github.com/MaslowCNC/Firmware/pull/152/files) which can be used to manually define the chain lengths (ex `B06 L1500 R1600 ` will set the length of the left chain to be 1500mm and the right chain to be 1600mm.

I don't think we should modify Ground Control to send this command because it will invariably cause chaos. @davidelang is right we should focus on less manual entering of values, not more to improve our calibration routine.

BarbourSmith, 2017-03-30:
Are we comfortable closing this issue?

jessbussert, 2017-03-30:
Yes.  Tyvm

closed 2017-03-30 by BarbourSmith