MaslowCNC/GroundControl #154: Separate setting for Z-axis move amount for Z-In and Z-Out

blurfl, 2017-03-29:
There is some interaction between the 'Distance to Move' setting and the distance the Z-axis motor travels for the z-in or z-out button. The relationship is not clear, and when large distances are set for horizontal traverse, the z travel will be large as well. It is possible to run the z-axis mechanism into its physical limits causing damage to the equipment.
 Having a separate setting for z-axis distance would solve the issue. It would not need to be as accessible as the horizontal distance setting, perhaps it could live on the settings page rather than on the front page.

BarbourSmith, 2017-03-29:
Good idea. Right now the relationship is 10:1 so the z-axis will move 1/10th as far as the XY axis. 

I think figuring out where to put that dialog is going to be the tricky part. What about replacing the the unused "test" button with a z-dist?

blurfl, 2017-03-29:
I understand now. I've been using large values to move the sled around the workspace, and when I go "Home" the automatic z kicks in. I'll unlatch the router before traversing.
 The "test" button would be fine, though putting it on the settings page would do too. I suppose it depends on how often one would use the 
'Test" function. I use it (from Actions) pretty often, trying to dial in some accuracy 😕 in the vertical direction.

BarbourSmith, 2017-04-18:
I broke everything related to the z-axis out into a popup. 

What do you think? Is it better? Am I missing anything?

blurfl, 2017-04-18:
I’ll try it out - too sounds like a good idea, though.

BarbourSmith, 2017-04-18:
Thanks, it wasn't really my idea tho :)

blurfl, 2017-04-18:
Those are all the z-things, conveniently grouped! Not sure how much you want tested, but here goes :)

- It takes a bit of figuring out - inches/mm shows the current setting, but switching between the two doesn't change the distance value as it does for the xy setting. Could changing the units keep the setting and show the value in the new units? 
-  I'm only able to use the 'in' and 'out' buttons once; after that the z-motor whines but does not move.
-  No matter what distance value I choose (.25 inches for instance) if I leave the popup with 'done' and come back the distance value is set to .1 ; 
-  Could 'In' and 'Out' be 'Z-In' and 'Z-Out' to avoid confusion about in and inches?

BarbourSmith, 2017-04-18:
Those are all fantastic points which are easily addressed. Thanks for the guidance!

BarbourSmith, 2017-04-18:
I've pushed a new version with the following changes:

'In' and 'Out' are now 'Z-In' and 'Z-Out'

Changing the units has the same behavior as in the XY view where the value is reflected in the new units

Closing the popup and opening it again does not change the distance set

What do you think?

Thanks again for the FANTASTIC suggestions.

BarbourSmith, 2017-04-18:
When you are happy, I am happy! Ok to close the issue?

blurfl, 2017-04-18:
Ok to close it :)

closed 2017-04-18 by blurfl