MaslowCNC/GroundControl #157: make this gcode run in GC

hannahebteagle, 2017-03-29:
hello! :) attached is the ZIPPED version of the .nc that won't open correctly yet in Ground Control :)))
[untitled folder 2.zip](https://github.com/MaslowCNC/GroundControl/files/880214/untitled.folder.2.zip)

BarbourSmith, 2017-04-01:
Fixed in #170

BarbourSmith, 2017-04-01:
There appear to be some scaling issues with the files, but that's not a Ground Control issue

BarbourSmith, 2017-04-10:
The latest changes appear to have made this code not work again, more investigation is needed

BarbourSmith, 2017-04-11:
First thing is this file doesn't specify units (G20/G21) or coordinate system (G90/G91) at the beginning which is poor form. Ground Control guesses that the units are the same as the last file opened, which isn't ideal, but there's not a lot else we can do about it.

BarbourSmith, 2017-04-11:
Fixed again in PR #193

closed 2017-04-11 by BarbourSmith