MaslowCNC/GroundControl #159: Rasp Pi Touchscreen screen size

blurfl, 2017-03-30:
The 7" Rasp. Pi Touchscreen cuts off the lower portion of the GroundControl panels. This leaves some actions and settings unreachable and the current lines of machine feedback off screen.

BarbourSmith, 2017-03-30:
I see a couple ways to resolve this. Either we make the buttons smaller, or we split the UI into multiple tabs or viewing areas.

I think this is an important issue because I'd like to make an Android App soon too and we're going to see all of these issues there also.

Kivy will let us scale buttons (and everything else) in one of two ways, it can either be scaled as a percentage of the screen size, or set to an absolute size. I went with absolute size because as a percentage of the screen the buttons looked too big on big screens and too small on little screens, but that means that if the screen size is less than the absolute size of the buttons, they get cut off.

If we decide to split things into tabs, I think the big question is what goes on in each view and how many are there. We can do separate viewing areas using either a spinner widget (built into kivy, works well) or something a little slicker like [this](https://www.youtube.com/watch?v=0nlUIbQrf5k). Kivy can give us the [window size ](https://groups.google.com/forum/#!topic/kivy-users/BNmZxiWd5Mc) which could be used to switch automatically between little mode and big mode.

What does everyone think? What's the best way to go about it?

BarbourSmith, 2017-03-30:
Possible duplicate of #144

blurfl, 2017-03-30:
I’d say that yes, this is the same issue. Go ahead and close the one you choose. Weed the garden!

BarbourSmith, 2017-03-30:
Sounds good

closed 2017-03-30 by BarbourSmith

jbarchuk, 2017-04-04:
> We can do separate viewing areas using either a spinner widget (built into kivy, works well) or something a little slicker like this.
Nooooooo horizontal-scroll ANYTHING!! EVER!!!  It may look cool and slick but is the BANE of UI. When's the last time you saw a MARQUEE tag used in HTML. Not in many years. When it was created everyone said Oh Wow Cool Let's Use That. And within weeks every page on the net had a MARQUEE. Until people figured out that nobody read anything in them because they were either too fast or too slow. Gradually they faded out. 
The problem is the TIME and effort it takes to find and select something. Well, first off, the user shouldn't -need- to 'find' anything, it should be right on the screen, tap it and go. With the horizontal scroller and the user needing to pick one of six items when only two at at a time are displayed the user needs to scroll, read, decide, 'nope,' scroll, read, decide, 'nope,' scroll more. Rather than a vertical list where all the words are displayed and the eye scans down to find the word they want to tap.
Just because an invention is clever and chic doesn't mean it's a good idea to use it.
The human eye doesn't read every letter of every word to find what it needs. It sees 'clumps' of letters, and the brain determines if enough of those letters are similar enough to the word it's looking for to take the time to look at more letters to find the exact word. There's a recent 'fun info' email going around, a fairly long paragraph made entire of scrambled words. It said that about 40% of people have no trouble reading the text at almost normal speed.
The brain is -faster- at decoding and making decisions than conscious cognitive comprehension.
Pleeeease don't slow down the whole process by doing anything with horizontal scrolls!!!

BarbourSmith, 2017-04-04:
Good suggestion. I agree that side scrolling is ugly and confusing. We have the option of vertical scrolling, but I don't think even vertical scrolling will feel natural on a touch screen mounted to the machine. It sounds like tabs is the way we want to go.