MaslowCNC/GroundControl #160: Rasp Pi Touchscreen keyboard types double characters

blurfl, 2017-03-30:
Using a keyboard to enter values in the settings produces double characters per keystroke. Typing  a '1' produces '11'. This makes entering values impractical.

TheRiflesSpiral, 2017-03-30:
Check the config.ini file for multiple or generic HID input types.

[input]
mouse = mouse
mtdev_%(name)s=probesysfs,provider=mtdev
**hid_%(name)s=probesysfs,provider=hidinput
%(name)s=probesysfs,provider=hidinput**

The last two lines in this file cause USB HID input to be handled by two providers. Try deleting the last line.

blurfl, 2017-03-30:
Once again, you've done it. 👍 Thanks for your help.
For posterity, on my (generic) kivypie setup. I edited 
/home/sysop/.kivy/config.ini and commented out the line:
%(name)s=probesysfs,provider=hidinput

Note that this change also corrected an issue with mouse-clicking which affected the mouse but not the touchscreen, causing double-clicks instead of single clicks.

TheRiflesSpiral, 2017-03-30:
I'll update the instructions to include this edit in KivyPi.

BarbourSmith, 2017-03-30:
Beautiful work everyone!

BarbourSmith, 2017-04-06:
Has this issue been resolved?

blurfl, 2017-04-06:
I remember some discussion about this after @theRifleSpiral posted the fix on the forum.I bumped the thread to see whether anyone thinks it should stay open. It certainly worked for me, “close” is my vote.

TheRiflesSpiral, 2017-04-07:
If you're running python under root (sudo python main.py from the ~/GroundControl directory) Kivy will use the root's config.ini file, not sysop. I updated the wiki to include an edit to both the sysop and root users.

BarbourSmith, 2017-04-07:
Fantastic! Thank you for updating the wiki to include both edits, and good sleuthing to figure out what was going on!

closed 2017-04-07 by BarbourSmith