MaslowCNC/GroundControl #161: Rasp Pi Mouse cursor click location misaligned

blurfl, 2017-03-30:
Using a mouse in GroundControl there is a blue dot for a cursor. The hot spot when the mouse is clicked is above and to the left of the cursor image.
 A comment on the forum suggests that there might be "a folder that holds the image files that appear during touches and when the cursor moves… "

TheRiflesSpiral, 2017-03-30:
The doc describing the touchring module is here: https://kivy.org/docs/api-kivy.modules.touchring.html

TheRiflesSpiral, 2017-03-30:
Here's a solution... It's a bit of a hack: replace the image "defaulttheme-0.png" in your [Kivy]/data/images folder with the image below.

[image]

This will put a fancy Maslow mouse cursor that is more obvious where the click will happen.

If you're just looking for a simple mouse cursor, use this file: 

[image]

BarbourSmith, 2017-03-30:
Is there a way we can make this easier for everyone? Possibly put this file into the Ground Control GitHub repo so that it can be moved into the proper location with a single command during the setup process?

Love the Maslow mouse cursor 😀 

Great fix.

TheRiflesSpiral, 2017-03-30:
Absolutely. I'll work on that after lunch.

blurfl, 2017-03-30:
That's the bunny! Thank you @TheRiflesSpiral 💯 ! I like your Maslow cursor :) 👍 
For posterity, on the kivypie image (mine, but I think it's generic) the location to put the defaulttheme-0.png file is:
/usr/local/lib/python2.7/dist-packages/kivy/data/images

BarbourSmith, 2017-04-06:
Has this issue been resolved?

blurfl, 2017-04-06:
I would say yes.

closed 2017-04-11 by blurfl