MaslowCNC/GroundControl #165: Empty increment size crashes GroundControl

GeroBH, 2017-03-31:
On
Ubuntu 14.04.5 LTS trusty
Python 2.7.6

Clicking "increment size of movement" opens with blank value, no cancel button, clicking "done" without entering anything replaces current increment with "blank". Clicking on "units" afterwards makes Ground Control crash.

File "/home/ghadmin/Maslow/GroundControl/GroundControl-master/UIElements/frontPage.py", line 94, in onUnitsSwitch
     self.moveDistInput.text = str(float(self.moveDistInput.text)/25)
 ValueError: could not convert string to float:

BarbourSmith, 2017-04-01:
I can replicate this on windows. 

Thank you for spotting that. Should be a pretty easy fix. 

I think the real question is what is the right behavior if you click "done" without typing anything. I'd say we should just leave the value what it was before the popup was opened.

BarbourSmith, 2017-04-01:
Fixed in #165

closed 2017-04-01 by BarbourSmith