MaslowCNC/GroundControl #166: Define Origin changes mm to inches

GeroBH, 2017-03-31:
On
Ubuntu 14.04.5 LTS trusty
Python 2.7.6

Clicking "Define Origin" changes mm to inches.

BarbourSmith, 2017-04-01:
Good catch!

This one is a little tricky. What's happening is that "define origin" is reloading the gcode file and setting the units to the units of the file. If the file is saved in mm that's mm, if it's in inches its inches.

I think that the real issue is maybe that you can switch units with a file open? Like if I open a file in inches, switch to mm what does that really mean? As soon as I click "run" the machine will switch back to inches mode (otherwise the part would be huge), but the units will display incorrectly (I believe).

So the real question is what is the right behavior for the situation where you open a file with units of inches, switch the machine to mm, then press "Run"?

GeroBH, 2017-04-02:
It is not an issue then, it's correct behaviour and can be closed. I think we can fairly assume that a user who prefers metric, will have his CAD/CAM setup in mm. Most sharing of designs will be on CAD level. If a user wants to use a .nc file shared with him, he should accept any unit it is created with. -> Joke: Unless you want to program a real-time unit conversion into GrondContol :-)

closed 2017-04-02 by GeroBH