MaslowCNC/GroundControl #171: Gcode reloads to 0,0

blsteinhauer88, 2017-04-01:
I am cutting projects with the final Maslow.  I move the cutter to my start location and define zero.  I load the Gcode and it centers on the cutter bulls eye.  Great.  I have had to update my Gcode, say for Cutting depth, or I put in the Material Thickness, or Speed.  I press hold on the project, then stop and return to Home.  I re-load the Gcode, and ground control puts it at 0,0 center, not where I defined zero.  Then if I push Define Zero again, it locks up, or it puts the pattern no where neat my Zero point.  Can it be revised to LOAD the Gcode to "My Zero". OR even slicker...If I hold the cut, make a change, reload the Gcode, press run and be off and cutting with the new code from where I left off?  At least have it reload to "My defined zero".

BarbourSmith, 2017-04-01:
Yes! Great, clear explanation of the desired behavior.

The two steps are 

1) Load gcode in target position.
2) Don't reset the gcode index so that it resumes from the same line.

Those are both things we can do.

One warning, if you change the gcode and update it, starting from the same line doesn't guarantee that you will start the cut in the same place. For example, if you remove a cut from the program or change the number of passes, you could end up starting from the same gcode line number, but a different place.

To make that more clear I think that the best behavior is that the g-code will always load in the same place and you can click "Run" to start from the beginning, or "Continue" to pick up where you left off.

Great suggestion!

blsteinhauer88, 2017-04-02:
Yes, I figured out the code line numbers as I watched it cut. I learned the advance 10 lines and watched it go back to where I left off. It had 4000 or better lines so I pushed >10 several times.  Ha ha, but got it back and continued.

BarbourSmith, 2017-04-04:
Follow up notes to myself about this:

I've been looking into it in [this branch](https://github.com/MaslowCNC/GroundControl/tree/fix-for-171)

Not clearing the gcode line number was easy. Now if you reload the gcode the line number is preserved. Re-applying the transform is a little trickier. I think that the desired behavior is that if the file name changes, it resets to zero, if you reload the same file, it comes back in the same place. The trick is to apply the transformation in the right place. The way the code is set up is that I can trigger the translation if the gcode changes AND the file address doesn't, the problem with that is that when it applies the transformation it causes the gcode to change without changing the file name and so it keeps looping.

BarbourSmith, 2017-04-04:
The more I look into it, the more I think this is a great idea. I've been in the position that this would solve many times.

BarbourSmith, 2017-04-06:
I think I've got a complete fix for this in pull request #175 , when you get a chance, take a look at it and see if there are any other tweaks to the behavior that could make it better. 

Great suggestion. I ended up re-writing a lot of the way files are handled, and the program is much the better for it.

blsteinhauer88, 2017-04-07:
OH Yea, that works great!! I used to get a "not responding" window when loading the nc file, or when pressing define origin...I love how it redraws the nc code on the screen, and no hang ups!!!

BarbourSmith, 2017-04-07:
Woo!!! Fantastic. Thanks again for the suggestion!

closed 2017-04-07 by BarbourSmith