MaslowCNC/GroundControl #172: Runaway Z-Axis

blurfl, 2017-04-03:
For the second time since enabling the z-drive, I've seen it begin a depth change and fail to stop at the desired depth. The z-axis motor continued to drive until its connector was pulled. Both times the x- and y- drives activated for the following command, following the correct path, as if the z change had completed.
 Is there a way to enable a log to capture info that would help? The issue today occurred executing line 10 of the attached gcode file.
[Big Grid Shifted2.txt](https://github.com/MaslowCNC/GroundControl/files/888836/Big.Grid.Shifted2.txt)

BarbourSmith, 2017-04-03:
Interesting. It doesn't seem to be an every time issue because when I ran the code just now it behaved normally so we're looking for an intermittent problem. I could see if the connector in the z-axis motor wasn't making good contact with the motor it could still get power but disconnect the z-axis motor and stop reading the encoder. The way to check that would be to see if the values are still incriminating on the digital read out.

The other question I would look for would be what was the last line of code sent to the machine. If maybe somehow the line got corrupted and dropped the decimal the machine would have received G1 Z-010000 F9.0 which would cause the expected overshoot.

blurfl, 2017-04-03:
Thinking further, the hardware cause seems more likely. Have there been instances of runaway X- or Y- travel? If not, then more likely the loose connector I think.

blurfl, 2017-04-03:
I have the z cable bound to the router power cable with a bit of a strain loop, but I'll cable tie it to the motor body to make sure. 
 The sled was in the extreme lower left of the work area, beginning to plot the second time around a 12" square. I don't have the measurements handy but it was within a couple mm of 12x12. Not quite square, but close. In the morning I'll patch the paper, tie the cable down and run it all again.

BarbourSmith, 2017-04-03:
I might be way off about the loose cable idea, that's just the only thing I can think of off the top of my head. Let me know if you see any other strange behavior that could shed more light on what is happening.

blurfl, 2017-04-03:
Is there any way to do some logging in GroundControl? I didn't see anything, but hope springs eternal...

BarbourSmith, 2017-04-03:
There isn't right now, but that would be a good idea. At the very least we can record everything that goes on on the terminal window. Getting more information out of the firmware is possible, but I usually delete any debugging lines as soon as I've tracked down the bug because they end up cluttering the serial communications

blurfl, 2017-04-03:
If it comes up again, might be worth setting up. For now, I'll tie down that cable.

closed 2017-04-11 by blurfl