MaslowCNC/GroundControl #177: Return to Origin Button

blsteinhauer88, 2017-04-07:
The Home button now returns the sled to 0,0.  It used to return to Where you set the Origin.  I like the Home returning to 0,0, However.  I saw today somehow my Gcode skipped several steps?  Wierd, cant say it was not my fault.  SO, I make a mark or record my XY at my origin before cutting just in case I have to return to start over, or redo a section.  If there were also an option/button to "return to where you set origin" It would take less time to get back to start over, or redo some steps.  What you think?

blsteinhauer88, 2017-04-07:
Playing more with this....I went "HOME".  I creaded from the same Makercam file just the outline because Ground Control got lost on last cut.  I  of the outline.  I wanted to restart just the outline.  I loaded just the Gcode outline into Ground Control.  You fixed so it would load to "Last Origin".  I found even though I pressed Run from Home...0,0, It traveled to the exact spot I needed.  I think this may be sufficient but would have to be explained how to do in instructions.

blsteinhauer88, 2017-04-07:
[image]
Look at where it just took off on its own???

BarbourSmith, 2017-04-07:
The taking off on it's own is very strange. Did it do that on the first pass?

I know exactly what is going on with the "Home"  button taking it to (0,0) always, I messed that up when I changed how files are loaded, but it should be an easy fix. The "Home" behavior I think is correct is that clicking **Home** takes it back to where the origin was defined, clicking **Action -> Return to Center** will take it back to (0,0). Does that seem like the target behavior?

Going rogue like that seems to me like a separate (but quite possibly related) issue. I'm less sure of what would cause that, let me think on it a little.

blsteinhauer88, 2017-04-07:
Yes I agee with leaving the 0,0 to the Retun to Center.  If you can fix the "Home" to return to Origin, Awsome!  So this project was wierd.  It loaded, I moved the origin a couple times, maybe 3, then hit RUN.  It actually started with some of the stars.....I did not know if that was good or not, just watched it run for a while.  It did the top half of the stars, then started the outline.  Now, I know this is wrong because the stars and stripes where first, shallow cuts, then the outline was through cuts.  

I stopped it, re centered it, and ran again, paying attention to the step numbers.  It started at 0 and went to the stripes first!! It skipped this last time.   let it run,  I walked away for a bite, then heard it cutting different....I returned to see it arcing across the wood.   Then I reloaded the outline and finished the cut. 

So, I am not sure what pass it was on, on the second try, but by the depth it appears like the second cut of of 4 steps.

BarbourSmith, 2017-04-08:
It sounds like I've got to spend more time actually cutting things to keep an eye on those weird bugs that pop up. I've been pretty deep into software for the last two weeks and sometimes something that seems to work perfectly in a quick test acts strangely when really making things. I'll keep an eye on it.

BarbourSmith, 2017-04-10:
Added in #185

closed 2017-04-10 by BarbourSmith