MaslowCNC/GroundControl #180: Limit opening files to type .nc

BarbourSmith, 2017-04-09:
Right now GC will let you try to open any file type. Limiting the program to open only .nc files with a message explaining what a .nc file is would be more clear.

davidelang, 2017-04-09:
at the very least, make it a configurable list of file extensions, not every app 
uses .nc as the extension for gcode files.

On Sat, 8 Apr 2017, BarbourSmith wrote:

> Right now GC will let you try to open any file type. Limiting the program to open only .nc files with a message explaining what a .nc file is would be more clear.

BarbourSmith, 2017-04-09:
That was actually the reason I didnt limit it originally. What are some of the others you have seen?  I've seen .nc .txt and .gcode

davidelang, 2017-04-09:
I've also seen .ngc, and files with no extension at all

There is no standardized extension for gcode files, and they are just text 
files, so people end up using whatever they want.

As an enhancement, a pulldown to adjust what files are shown (including all) is 
not a bad idea. Ideally GroundControl should remember the user's setting for 
this from run to run.

BarbourSmith, 2017-04-09:
Good suggestion. I'll include .ngc also by default, but also make it a user option stored in the settings file.

jbarchuk, 2017-04-09:
David Lang
> As an enhancement, a pulldown to adjust what files are shown (including
all) is
> not a bad idea.

That's the most common and efficient way to do this. The default is nc. In
a dropdown list the other common known extensions are listed. And 'All
files' at the bottom of the list.

The point is to get the user to their file as fast as possible.

What I've never seen done but should be possible is a user-definable
default (the one that appears at the top of the dropdown.) Let's say
someone users ONLY .gcode and NEVER anything else. Let the user put that at
the top of the list and it saves the user a THOUSANDS OF CLICKS a year.
Users luuuuv software that helps them do less work. :) OK never mind
really, that's a frill feature. It'd be cool though to be the first to do
it and try to make the idea catch on.

BarbourSmith, 2017-04-11:
Resorting the files based on extension is a great idea. It looked like it was going to be tough to do, so I stuck to just adding a user definable list of extensions which are accepted. If the user tries to open another type of file, the current file remains open and a popup explains the problem. Added in PR #188

closed 2017-04-11 by BarbourSmith