MaslowCNC/GroundControl #181: Make a button to define chain lengths manually

BarbourSmith, 2017-04-09:
Running the chain length calibration process over and over can be tiresome. It would be nice to have a button in GC which you can click when the sled is centered which will re-calibrate the chain lengths.

jbarchuk, 2017-04-09:
I like this idea a lot for people who will be using a piece of random scraps of non-4x8 material. Maybe even never using a full 4x8 sheet but having the store cut it in half, bringing home 2 4x4 sheets, and mounting both sheets on the frame at the same time. Being able to quickly tell the GC where the material is located on the frame without a long involved calibration process is a major time saver. ANY improvement that gains time is a MAJOR plus.
Given that no two pieces of plywood are the -accurate- or the -same- in ANY dimension, the repeatability you mentioned in your other post is rather moot. Repeatibility will be a matter of the user placing the working material on the frame such that it matches as closely as possible the location of the first 'calibrated' material.
(Difficulties will arise when the user tries to work too close to the edge, and miss the mounting of the material in the frame by --><-- that much. Maslow hardware and software can't allow for that, so the user must be cautioned.)
Given those caveats I think this kind of calibration -can- be done by eyeball down to 1/2-1mm accuracy.
Finding the center of a rectangle is a matter of drawing two diagonal lines from opposite corners. That sounds easy but very few people have an -accurate- 9' straightedge. But I can think of another way...
Tap a tiny brad/nail in the four corners. Tie two strings that connect the opposite corners. Where the strings intersect draw a tiny dot. (And draw a circle around that so you don't 'lose' the dot in the expanse of wood. LOL!)
I wonder if also calibrating the 4 corners would help the software figure out where things start to get 'less than optimally accurate?'
But here's another technique, no straightedge or center. The calibration could tell the GC where the 4 -corners- are, and the GC automatically knows where the center is. (Actually the GC needs to know only where 3 corners are, but 4 might improve accuracy.)
The 'pointing' device mounted in the router could be a lot of different things, a dowel or piano wire sanded to a point, or a nail with the head cut off.
Process...
Whether manual or 'z-axis adjustable,' the user sets that pointer to about 1mm off the working surface.
Eyeball the pointer from all sides as it hovers over a target point.
Use the control panel to fine tune the pointer to match the target point.
Eyeball the point AGAIN from all sides.
Click 'done.'
This sounds rather painless. Then again I know it's easier for -me- because I've been doing these kinds of things for a very longtime and know how accurately my eyeball is/isn't calibrated. It'd take a bit of a detailed process description to make it doable for the 'not incredibly mechanically inclined' user.
As always, if ANY user has ANY problem, difficulty, or confusion, please say so. If -you're- having a problem then the chances are about (approximately) 100% ;) that someone -else- is having the same problem. Every problem that's fixed is amplified by 10 or 30 other users.

BarbourSmith, 2017-04-11:
I added a manual calibration button next to the automatic calibration button which manually defines the chain lengths to be 1900mm long. Because every frame is a different size, this might not be the exact center. Added in PR#189

closed 2017-04-11 by BarbourSmith