MaslowCNC/GroundControl #197: 'Test feedback system' sends sled to Y-30.7218

blurfl, 2017-04-13:
Not sure why; looking at the gcode, it should have been Y-0.7218. The machine interface window indicated Y-3/0.7218... I haven't captured the steps I used to cause this; I'll try to note them down.

BarbourSmith, 2017-04-13:
Hmmm I don't know what the cause of that would be off the top of my head. Does it do it every time? Very strange.

blurfl, 2017-04-13:
Did it twice in a row. FWIW, it showed the test figure way down below the work area rectangle on the screen.
 I redid the auto cal, did a few 'homes' and loaded a gcode program and ran that. Then I went in and the test ran correctly, with a -very- nice low 0.16mm reading! Got 0.87mm immediately after that. Router not running, don't know if that matters...
I wonder if this has something to do with the various grbl coordinate frames? GroundControl was adding -30 to the Y coordinate.

BarbourSmith, 2017-04-13:
Hmm when the test is loaded it applies the same "Move Origin" shift hat was applied to the last file which was open, so it could be that just ended up as -30. Does that fit with what you were seeing?

.16mm error is fantastic. I've been getting in the .8mm - 1mm range consistently. I mostly added the test to show how much room we have to improve in the feedback system (it's a lot). I'd rather have room to improve than be at the limits of what's possible.

blurfl, 2017-04-13:
It must have been the "Move Origin" then. Not sure how it got set, but that explains it.

closed 2017-04-13 by blurfl