MaslowCNC/GroundControl #200: NC file too big to load due to "Follow the Path" lettering

blsteinhauer88, 2017-04-14:
Here is the file that would not load.
[Bench Back For Kids (2).zip](https://github.com/MaslowCNC/GroundControl/files/923125/Bench.Back.For.Kids.2.zip)

BarbourSmith, 2017-04-16:
I'll dig into it and see what I can find out! Thank you for sharing it.

BarbourSmith, 2017-04-17:
@blsteinhauer88 Are you sure that's the one that doesn't load? That one looks to me like it's just the outline, but it seems to load ok

[image]

blsteinhauer88, 2017-04-17:
[Bench Back For Girls Final.zip](https://github.com/MaslowCNC/GroundControl/files/926406/Bench.Back.For.Girls.Final.zip)
Try This one, big!

BarbourSmith, 2017-04-17:
That looks like the one :-)

I'm seeing 50,000+ lines, and it doesn't load

BarbourSmith, 2017-04-17:
This one is going to be a little tricky. I've got figure out what's slowing everything down and it seems to have to do with the way the underlying graphics system is processing everything. 

As a note to my future self, the first thing to look into is to if rendering the whole thing as one polyline iwth 50,000+ points will render faster than 50,000 individual lines. 

I've added an error popup, and at least the file will try to load now, but on my computer GC gets VERY slow. Added in #205

BarbourSmith, 2017-05-13:
Support for big files (up to 60,000 lines) added in #279 

[image]

Thanks for suggesting the change, sorry it took me so long to make it. 

Good to close?

blsteinhauer88, 2017-05-14:
Yes looks good!

closed 2017-05-14 by BarbourSmith