MaslowCNC/GroundControl #201: Crashing when clicking "Home"

BarbourSmith, 2017-04-16:
I've seen an issue and heard it mentioned in the forums that clicking the **Home** button on this week's build can cause GC to crash. I've had a hard time making the problem occur repeatably in a way that I can track it down, but here are the facts that I have found so far.

1) It seems to happen only in the compiled version which means that the error message flashes on the screen too fast for me to read what it says.
2) Opening the settings and changing the z-axis from enabled to disabled and back again seems to fix the issue?

Has anyone else seen this problem? Does anyone have any idea what the error message says when the program crashes?

jbarchuk, 2017-04-17:
> the error message flashes on the screen too fast

Set up a video camera to record the error. Or screencap software, but that -might- introduce timing that causes the problem to -not- happen.

BarbourSmith, 2017-04-17:
I tried the video camera idea, but I still couldn't quite get it. I think the issue was that my phone's frame rate is too slow. Then I realized that I could just put a pause command in the .bat file which launches the program so after it crashed it would hold the error open. Here's the message, an easy fix.
[image]

BarbourSmith, 2017-04-17:
Got it in PR #203

closed 2017-04-17 by BarbourSmith