MaslowCNC/GroundControl #202: Loss of position when cutting files from Fusion 360

BarbourSmith, 2017-04-16:
Cutting files from fusion 360 seems to be less reliable than other methods of generating G-code probably because I haven't tested it as thoroughly, but making it work is a priority.

nwrancher, 2017-04-16:
[Letter Box NC.zip](https://github.com/MaslowCNC/GroundControl/files/924493/Letter.Box.NC.zip)
Here are my most recent nc attempts, at 5ips.

nwrancher, 2017-04-16:
Using this link you can edit and output yourself from Fusion360, if you'd like to play with the Z settings, or turn on/off G28, or any of the other variables.

http://a360.co/2oNiMP8

BarbourSmith, 2017-04-17:
@nwrancher I am seeing the same strange behavior you are and I'm working on it right now.

BarbourSmith, 2017-04-17:
I believe it should be fixed in PR [#176](https://github.com/MaslowCNC/Firmware/pull/176). 

I looked into the strange z-axis behavior you were seeing and I think there are two causes. 

1) The gcode is telling it to do some weird stuff. The first few lines of code are:
`G0 X0.2061 Y1.8705`
`Z0.3`
`Z0.2`
`G1 Z0.1594 F20`
`Z-0.1075`

Which command the machine to move to the starting position (no z-axis change), then move as fast as possible to heights of .3 inches, down to .2 inches, then at speed 20 inches per minute to height of .1594 inches, then again move down to a depth of -.1075in. None of this is really an issue, it just probably looks weird. 

If you hare having issues with the bit cutting the surface before it gets to the starting position, you can always bump it up by clicking the "Z-Axis Out" button in GC before clicking "Run" because the program doesn't change z-axis position until after it arrives at the start this will insure that it doesn't touch the surface.

2) The second reason you might see strange behavior is that Fusion 360 seems to like to throw in lines like `G19 G3 Y1.8705 Z-0.1075 K0.0125` which use the G19 command to switch to the YZ plane of reference. It only does it rarely and they never seem to switch for more than one line, but it could still cause strange behavior because we don't support the YZ coordinate plane right now and it's a pretty big change to add it. I will talk to them and work on getting rid of those.

blsteinhauer88, 2017-04-17:
Question, once "run" is initiated...Can the Z in or out be used or will it mess with the running code.  If not?, if the 'Hold' button is initiated, can then the Z in or out, or even the manual movement arrows, be used, then the 'continue' restart the running of the code?

nwrancher, 2017-04-17:
That's all helpful Bar, thank you.  The z stuff is fusion and their retract height, clearance height, feed height, and then ramp feed settings.  I still believe I'm having some problem prior to the start of the file, with GC's safety height, or something.  It will all look great, and the code reads as it should, but it all happens half an inch above the surface.  And, things go by so fast in GC I can't see why.  May I suggest replacing the >> buttons with a gcode line by line button.  Next line/last line, or something.  I don't know if that's a good idea, but it might make things easier to diagnose.  I do know that I see Z 0.25 right at the beginning and that's not part of the code.  But then things start going by too quickly.

nwrancher, 2017-04-17:
I should also add that the latest iteration of that "riser" file did exhibit the G19 behavior, although I had no idea what was going on.  That was last week.  It did the first pass great, then the second one all the inside curves were outside and the x was y.  I could not figure out what I did wrong with that one!  Good to know it wasn't me.

BarbourSmith, 2017-04-17:
I like the idea of having the ability to step through the program one line at a time. I don't want to replace the existing >> buttons with something new because I think that the existing behavior of being able to pause and then jump forward in the program is useful. I think it's a great idea, and you should make an issue for it. If you have any ideas about how to make it look on the UI, I would love your suggestions. 

@blsteinhauer88 great question. If you click the arrow keys or z-up/down while running a program the machine will respond to your command, then go back to running the program which probably won't have the results you want. Because G-code often relies on information from the last line you might get strange behavior, especially in circles. 

If you say wanted to pause the machine and adjust where the z-axis depth is defined you can do that no problem. Just keep in mind that the z-axis won't go back to the depth you want it at until it gets to the next command with a 'Z' in it in the program so you probably want to pause, re-zero the z axis, then lower it back down, then press resume.

nwrancher, 2017-04-18:
Bar, should I test that Letterbox file again?  Were you tracking after this fix?  I can test today, but I wasn't clear on your results.

nwrancher, 2017-04-18:
To be more clear, I'm asking about the position tracking, not the Z stuff.

BarbourSmith, 2017-04-18:
I believe that the position tracking should be better now. If you were to test it and let me know the results that would be very helpful. Thanks again for finding this issue.

nwrancher, 2017-04-18:
It appears to be tracking fine on the files I had problems with yesterday.  Thank you Bar!

BarbourSmith, 2017-04-18:
Thank YOU for finding the problem and giving me test files to track it down!

closed 2017-04-18 by BarbourSmith

BarbourSmith, 2017-04-18:
@nwrancher Just got finished meeting with the guys from Fusion 360.

They are going to add a Maslow post processor option with exactly the settings we want. Thank you for pushing this. I knew what we wanted to ask for because I've been digging into the .nc files you've been sending me. Thank you.