MaslowCNC/GroundControl #204: Ground Control Holding Position information.

blsteinhauer88, 2017-04-17:
After cutting a project, I return to center and 'zero' out the maching.  I then try and drive the sled to the new start position.  I will set the distance to travel variable, then say hit the left button.  The sled travels right!  Then stops at a random, or the last time I drove it manually, it goes that distance.  I then drive it again to the position I want it to start and his 'define origin'.  Then after a run command to make sure it goes to the start point, it travels in a random general direction, then the error ring shows very large!  As the sled travels it gets smaller and the target cursor finally lands on the start point, and sometimes it moves, but the sled it still.  I then try and return to center function, and or unplug and reload.  I thing GC is holding some old information.  Is there a way to after your finish a project/cut, you can clear the instructions.  Or when hitting 'return to center' it clears any buffers of information and zeros the machine.  Sometimes the Z axis wont retract on a Home or Return to Center.  Lost a coupe bits as it dragged.  Very random.

Also had a couple times where it just loses it's position and starts driving across the page.  I have the code for the last time it did this.  I will send it as I have not tampered with the final .NC nor re-cut the part that was destroyed.  The last pass cut from the lower left corner of the part, cut through at a similar arch as the part, but into the lettering in the name.
[Bench Back Brace.zip](https://github.com/MaslowCNC/GroundControl/files/926443/Bench.Back.Brace.zip)

BarbourSmith, 2017-04-17:
I think this might be a couple different issues manifesting as one, but I think I know a good place to start.

Right now the communication between Ground Control and the machine is always done in absolute coordinates so the command G01 X10.0 Y10.0 means go to the position (10,10) so to make the arrow buttons work Ground Control has to calculate the machine's position and add the distance to move to it which can lead to strange behavior if Ground Control gets the position wrong. 

The alternative is to switch to relative coordinates so G01 X10.0 Y10.0 means move 10.0 in each direction from where the machine is now. That way Ground Control doesn't need to know anything so there is not information kept in Ground Control to clear.

BarbourSmith, 2017-04-17:
PR #207 switches the way the arrow buttons behave such that they now use the relative coordinate system so Ground Control is not storing any position information anymore. This should give more consistent results. 

The exception is when the machine is first powered on and it doesn't know it's own position. In this case you will still see the large error circle until the machine catches up with where it wants to be.

I think we should leave this issue open because the file where it went rogue is probably a different error that I want to keep looking into.

BarbourSmith, 2018-01-10:
I think all the issues here have long since been fixed 👍 👍

closed 2018-01-10 by BarbourSmith