MaslowCNC/GroundControl #212: Retract bit, Pause, and Prompt for Tool Changes?

mindeye, 2017-04-19:
It doesn't seem like GC pauses at all for tool changes currently, which is fine when it's a false tool change but if I actually start milling things that need a tool change it'd be nice to have a chance to actually change the tool without manually hitting the pause button right on cue.

Note: this might be user error in that I'm missing some setting in the software but better bug incorrectly filed than missed.

BarbourSmith, 2017-04-19:
That's a great suggestion, and will probably be relatively easy to implement. Do you by any chance have an example of a file showing a tool change the way you want to see it work?

mindeye, 2017-04-19:
I can/will upload one tonight, but you can easily create one by just doing two profiles operations on the same shape with different tool sizes in maker cam. I have yet to actually _need_ this functionality but it's something I thought of while seeing my cleanup pass with a "different" tool.

mindeye, 2017-04-19:
It's probably also worth pointing out that we should ask the user to either confirm it's not a real tool change to quickly bypass it or re-zero their z-axis (and maybe move it back out to the safety height?) during a real tool change, otherwise the depth will be off between the two tools. I'm not certain what the best way to handle this will be in practice.

BarbourSmith, 2017-04-19:
Very cool. That's an easy thing to test.

I'm imagining spawning the same popup as for manually adjusting the z-axis so the user has the option to press "Continue" and the program will just keep running, or press "Hold" which will pause the program and let the user change the bit and re-zero the z-axis for the new tool. Does that seem inline with what you are looking for?

mindeye, 2017-04-19:
Yeah, that'd work great. I'm just wondering if it would be good to throw some disclaimer in there warning the user that they may need to re-zero their z-axis and put it back to the same position after swapping tools. I intuitively grasp that but I'm not sure that every user will... it's hard to decide when to assume 'expert' users and when to assume normal (maybe not always thinking things through) users. I know I still fall in the latter bucket on a regular basis :)

BarbourSmith, 2017-04-19:
That's a good point. I'll make sure the text makes it clear that the length of the bit may have changed and need to be re-zeroed

skiz, 2017-11-20:
Manual changes, especially mixed with other features like auto spindle control could be bad.  I would suggest that unless you have come up with an automatic tool changer, to run multiple gcode files after bit changes and zeroing.

mindeye, 2017-11-20:
Fair enough. Seems like a reasonable way to run things. Feel free to close
this issue if you like.

closed 2018-01-10 by BarbourSmith