MaslowCNC/GroundControl #223: Feedrate for files in mm is not working

GeroBH, 2017-04-23:
I am trying to cut a 100x100mm square. If I put F500 or F1500, makes no difference. The sled moves 0.5mm per second. A 6mm cutter. Something wrong with my handwritten G-Code?

(Generated by Hand)

G21 G90 G40

(profile 1)
G0 Z5
T0 M6
G17
M3
G0 X-3 Y103
G1 Z-2 F300
G1 X-3 Y-3 F1500
G1 X103 Y-3 F1500
G1 X103 Y103 F1500
G1 X-3 Y103 F1500
Z-4 F300
G0 X-3 Y103 F1500
G1 Z-2 F300 F1500
G1 X-3 Y-3 F1500
G1 X103 Y-3
G1 X103 Y103 F1500
G1 X-3 Y103 F1500
G0 Z5
M5
M30

GeroBH, 2017-04-23:
The good news is that it is square with an unbelievable precision. Within a 10th of an mm is amazing. I have shortened the file to do only one depth.

(Generated by Hand)

G21 G90 G40

(profile 1)
G0 Z5
T0 M6
G17
M3
G0 X-3 Y103
G1 Z-2 F300
G1 X-3 Y-3 F1500
G1 X103 Y-3 F1500
G1 X103 Y103 F1500
G1 X-3 Y103 F1500
Z-4 F300
G0 X-3 Y103 F1500
G1 Z-2 F300 F1500
G1 X-3 Y-3 F1500
G1 X103 Y-3
G1 X103 Y103 F1500
G1 X-3 Y103 F1500
G0 Z5
M5
M30

blurfl, 2017-04-23:
Bar has the maximum feed rate limited to 20ips, 508mm per second because of the need for more work in one section of the firmware. One could look into the firmware and alter the speed limit :)

blurfl, 2017-04-23:
Great news about the accuracy! What angle are you using? How is the accuracy when you move out fron the center?

davidelang, 2017-04-23:
hmm, thinking out loud, the maslow has a designed feed rate of ~48 ipm, if it's 
checking the value of the F command before it does the conversion, it could be 
taking your 500/1500/etc and cutting it down to 25 and then imposing the 
conversion factor of 25.4, resulting in the incredibly slow 0.5mm/s (which 
becomes 30 mm/min, which is suspiciously close to the 25 ipm limit)

David Lang

On Sun, 23 Apr 2017, GeroBH wrote:

> I am trying to cut a 100x100mm square. If I put F500 or F1500, makes no difference. The sled moves 0.5mm per second. A 6mm cutter. Something wrong with my handwritten G-Code?

GeroBH, 2017-04-23:
Just bumped it up to F10000, no difference. Same lame move.

blurfl, 2017-04-23:
I think you've put your finger on it, David. In CNC_Functions.h it looks like just what you describe is happening.

blurfl, 2017-04-23:
Gero,
 If you wanted to experiment, you could edit CNC_Functions.h in the Firmware and comment out line 410 by putting a '#' at the start:
#     feedrate   = constrain(feedrate, 1, 25*_inchesToMMConversion);  
Then upload that firmware and see whether the speed changes. Not the fix, but a test and maybe a bandaid :)

GeroBH, 2017-04-23:
Will do happy. Cut upper left, 0.5mm to small. cutting lower left now. After that I will give it a try. Thanks.

GeroBH, 2017-04-23:
That did it :-), good I pulled the cutter out and just simulated. F500 seems to fast. Will circle in to see what speed I think might work. Amazing fast you guys are. Great job!

blurfl, 2017-04-23:
Good news, We'll have to unwind the logic that causes this, but you're cutting at a reasonable speed 👍 .
 Your testing is providing interesting and important information. What angle is your frame set to?

GeroBH, 2017-04-23:
Well the angle, was wondering. Since I hinged the frame and just set it out until my frame was flat on the surface, I never bothered. The app I just installed said 6 degree. But I don't really trust it. Will come up with a better way of measuring that.

blurfl, 2017-04-23:
Just that rough guess is good enough, thanks :) I've been using a bigger angle, but I'm going to try a smaller one - it seems to be working well for you...

jbarchuk, 2017-04-23:
These app thingies are pretty accurate. You can test it with a known flat
surface (a cast concrete floor,) a known length of wood, and a known height
to prop up the other end.

The stock angle is 15 deg off vertical. Depending on speed, material and
bit, too low an angle will encourage the bit to jump out. 6 sounds kinda
risky but maybe doable under the right conditions. We don't know all the
trial/error issues Bar went through with development to land up at 15.

blurfl, 2017-04-23:
Good points, we should be adding the info about up- down- or straight bit when we talk about all this. Many variables, much to consider.

davidelang, 2017-04-23:
don't worry too much about measuring the angle in degrees, if you tell us the 
dimensions of the triangle (bottom, angled part, vertical part), we'll calculate 
the angle for you.

good to hear that it works well at about 1/3 the stock angle.

GeroBH, 2017-04-23:
The 3 cuts on right up, right middle and down middle are more off then the left ones. And the not square any more. Something under 1 mm but I am wondering if the slow speed on the left added to accuracy. Now that I cut a square with a little more that a minute, seems the right time to turn the sheet around and do a series over the entire sheet. The way I had determined my angle was as follows: I put a rope in the holes of the sled mount and hang the sled. I moved up the holes until I had the sled almost vertical. I am in the 4th hole from the sled surface now. I adjusted my motor mounts to about that same distance to have no side force on the chains. Then I turned the car jack of the base until the sled was sitting flat, just tipping with the finger on top and down to see if the as a move. A 7kg router with the gravity far in the back. 15kg is the sled with everything mounted. What is the risk in going as close to vertical as possible?

BarbourSmith, 2017-04-24:
Great work everyone! Three weeks and you guys don't even need me anymore 😀 

Great work @davidelang and @blurfl for figuring out what was going on and finding a solution. I will fix the issue ASAP

In terms of the angle I played around a little with different angles over the weekend, and I think we can go quite a bit steeper and still get good results. I was cutting with just the temporary sled (no bricks) because I was working on the instructions and it still worked well. Using an older or more worn out bit seemed to cause it to want to pull out of the wood. @GeroBH you might want to try using half bricks because your router is so heavy. I can't think of any big risk by going closer to vertical as long as you aren't cutting something irreplaceable in case it doesn't work.

jbarchuk, 2017-04-24:
> I can't think of any big risk by going closer to vertical as long as you
aren't...

Or using an upcut bit, because that is trying to push the bit out of the
cut.

davidelang, 2017-04-25:
On Mon, 24 Apr 2017, jbarchuk wrote:

>> I can't think of any big risk by going closer to vertical as long as you
> aren't...
>
> Or using an upcut bit, because that is trying to push the bit out of the
> cut.

do we really need this much angle for an downcut but (which is what would push 
the bit out of the wood, upcut will pull it into the wood)?

I think the original 15 degree angle was a "this looks reasonable" decision as 
Bar was building the first machine, not somethig that was arrived at via 
experimentation.

blurfl, 2017-04-25:
I’m rigging to try 10 degrees with an upcut bit, but one thing I notice is that the sled isn’t as firmly placed against the work surface. Using chain holes higher above the sled worsens the chain angle at the sprocket. Rancher’s idea for a yoke motor mount could solve that, though.

jbarchuk, 2017-04-25:
On Mon, Apr 24, 2017 at 8:01 PM, David Lang <notifications@github.com>
wrote:

> do we really need this much angle for an downcut but (which is what would
> push
> the bit out of the wood, upcut will pull it into the wood)?
>
I got that backwards, editing now.

blurfl, 2017-04-27:
@GeroBH, Bar has just merged the fix for this in Firmware. The fix limits the feedrate to 635 mmps. He will need to work on other parts of the code to enable speeds above 635mmps. Does this take care of the slow-speed problem?

GeroBH, 2017-04-28:
@blurfl, I will not be able to test till Saturday morning. Had a strange 'Buffer Overflow' and 'connection lost', the sled went places it should not go. Want to replicate that without cutter and get more details, with the old Firmware/GC first, then check it with the new Firmware/GC. Since its a mm file, should show me also if the fix is working. Need to calibrate chains and so on, but think that is a good idea anyway with all new software.

BarbourSmith, 2017-04-29:
I'm going to declare this one fixed (thanks @blurfl!), we can re-open it if it crops it's head up again

closed 2017-04-29 by BarbourSmith

blurfl, 2017-04-30:
👍