MaslowCNC/GroundControl #225: Transport buttons not responding

nwrancher, 2017-04-23:
I am not able to use the transport < << >> > buttons for any useful function.  Sometimes they appear to make it skip ahead, but I don't understand how to manage it so it isn't dragging the bit.

I would like to see them greyed out when code is running, active when on hold or no code running.  Ideally they would scroll the code in the viewer with the active line highlighted and position indicator showing where on the cut, then "enter" to run starting on that line.  Or something.  I'm not entirely sure how they should work, but currently, for me, they don't seem to.

BarbourSmith, 2017-04-24:
Yes, I agree those buttons have been neglected while everything else changed and they don't really work anymore. We need to fix them.

How they work right now is that they jump ahead to either 1> or 10> lines of gcode. None of the skipped lines are run, so if the bit is supposed to retract during that skipped part it doesn't. There used to be an indicator of where the machine was going to skip to, but that doesn't show up anymore as described in issue #227 

What is the behavior we want to see when clicking the arrow buttons? Do we want it to run the intermediate lines (that wouldn't be much of a fast forward)?

It shouldn't be too hard to fix, but I'm not clear right now on what we want them to do.

blurfl, 2017-04-24:
I think it is more useful to skip lines, that's the only way we have to ignore part of a file and move on. I suppose adding a 'Skip/Run' button would offer both options, but more buttons :( ?

BarbourSmith, 2017-04-24:
That makes sense. I think the first step is to put back in the indicator of where we are skipping to as described in #227 . Then maybe we can get a better sense of what we want.

blurfl, 2017-04-24:
👍

jbarchuk, 2017-04-24:
More options = less restrictions. A user might want to review steps ahead,
go back to the point where it stopped, and continue from there. For example
'I'm not exactly sure where the router's gonna go from here, so I want to
check the path first. At this point I have the options of quitting, going
back/forth manually in the script (a shuttle in media editing terms) and
continuing forward from that point, or going back to where the GC remembers
the last point where the script was stopped.' When the user has been
thinking about things and moving the sled around, it's less error prone for
the GC to remember where it stopped rather than the user.

johnbenweber, 2017-05-14:
I think it might be more useful to skip to the next z-move than a specific number of line. When I've wanted to skip it was to skip over a whole feature or a depth pass and you don't necessarily know how many lines that is.

blurfl, 2017-05-14:
I've often used the step over lines, but also wished for a 'skip to next z'... Another button!

BarbourSmith, 2017-05-14:
I think that's a fantastic idea. I usually end up using the skip buttons only to move to the next z-move.

Rather than adding more buttons, I'd like to start working on making it so each gcode object can be clicked on so if you had a number of parts you could click on one and run just it.

Probably a couple weeks away from being able to add that tho.

BarbourSmith, 2017-05-14:
Do we feel good about closing this issue because the original problem is solved, and opening a new issue for 'Jump to next Z move' or something like that?

blurfl, 2017-05-14:
OK by me 👍

johnbenweber, 2017-05-17:
Well I had a go, see PR #282.

BarbourSmith, 2017-05-17:
Love it!

johnbenweber, 2017-05-17:
Be careful with this, I think I just fixed an obo issue but it needs to be well checked.

BarbourSmith, 2017-05-23:
I just merged #282 which replaces the jump 10 lines buttons with a jump to the next z-axis move. Great work @johnbenweber 

How does everyone feel about us taking this even further? As I was testing it I realized that I really just want to jump to the next time the z-axis retracts above the surface of the wood. As is it jumps to the next tab for me a lot of the time. 

I propose that the ideal behavior for these new buttons is to jump to the line where the the z-axis next retracts above the surface of the wood. That way pressing the "Resume" button will lift the bit from the wood, then move on to cutting the next part.

The "Hold" button should also become "resume" when the FF buttons are used to indicate that the file can be resumed, rather than run from the beginning.

What are everyone's thoughts?

davidelang, 2017-05-23:
what if you have a z-retract that doesn't break the surface of the wood?

what if you have a ramp up that passes the surface in the middle of the 
movement?

BarbourSmith, 2017-05-23:
Great questions. Moreover that would leave us vulnerable to other corner cases like those.

davidelang, 2017-05-23:
If a clear policy can be defined that can handle corner cases, it sounds like a 
reasonable thing to have, but something that only works most of the time can be 
worse than not having the feature.

BarbourSmith, 2017-05-23:
I agree 100%.

I think it would be a useful enough feature that I'd like to try to find a clear policy that eliminates corner cases.

Can anyone think of one?

One more corner case to consider: The surface of the wood doesn't need to be at height 0. You can define the bottom to be 0 and cut down towards it so the z-axis is always positive.

blurfl, 2017-05-23:
Jump to next tool change? You would have to put tool change lines ahead of each part in your file, but that's doable.

johnbenweber, 2017-05-23:
It's probably worth thinking about whether there is a good way to detect tabs. I can't think of a simple robust way off the top of my head.

@davidelang if you make the change in z threshold a bit bigger you could avoid ramps as those would be composed of a lot of very small z moves.

@blurfl skip to tool change would be easy to do but then you have to anticipate where you would want to skip to when generating your gcode which sounds inconvenient.

davidelang, 2017-05-23:
G0 x-500 y-500 z-25
G1 x500 y500 z25

will make one continual movement 1.4m long and move the z axis 2 during that 
movement. It doesn't break it down into smaller movements.

It's only curves that have to be broken down into many small movements.

johnbenweber, 2017-05-23:
@davidelang you're right, I hadn't thought of that.

I know @BarbourSmith has mentioned making the gcode viewer interactive, perhaps that would be better than trying to have an intelligent skip function?

BarbourSmith, 2017-05-23:
I'm kinda cheating here because I think a key part of making the gcode viewer interactive is to be able to detect individual gcode objects. I was hoping that by finding a robust way to detect the beginning and end of each object now for the "skip to end" function we're coming up with now we could more easily separate all the different cuts which are part of the gcode file.

davidelang, 2017-05-23:
I think you're dreaming here, I don't think there is a reliable way to detect 
different objects. The gcode can be in any order (I've seen generated g-code 
that cuts part of several objects before it finishes any of them for example)

BarbourSmith, 2017-05-23:
That's a great point about cutting parts of several objects before finishing any of them. I've seen that too and there's really nothing we could do about that.

Once we build CAM into Ground Control we can have some control, but until then it's probably a pip dream.

For now I'm going to close the issue and award full points to @johnbenweber 

Any objections?

blurfl, 2017-05-23:
💯 👍 👍

closed 2017-05-23 by BarbourSmith