MaslowCNC/GroundControl #232: change open file process to be more efficient

davidelang, 2017-04-29:
to open a file it first reads every file in the directory (and apparently rather inefficiently) and then displays every one of them, with no ability to sort/search them, or type a filename

My home directory has >1200 files in it, and having to go through these graphically to find a file is horribly slow

BarbourSmith, 2017-04-29:
I completely agree. I'm not sure what the solution is because the file chooser is built into Kivy so I didn't create it. The right solution could be to use the native file chooser in tkinter?

BarbourSmith, 2017-04-29:
Although that won't work on android anymore...

johnbenweber, 2017-05-06:
The native kivy filechooser can handle file extension filtering, see PR #265.

johnbenweber, 2017-05-14:
I think this issue can be closed.

closed 2018-01-10 by BarbourSmith