MaslowCNC/GroundControl #233: Lost connection and GC reconneting behaviour

GeroBH, 2017-04-29:
Sorry I can't come up with a better title, feel free to change it.
Also explaining it, is a bit difficult.
But what I have observed is a potential danger. I'll try.

After a 'buffer overflow' and a ' USB connection lost' popup, with out any response buy the user, the sled moves downward. I assume it moves to where GC thinks the centre is. However, that is off after loosing connection, and a cutting bit should not move at all, after that, in my opinion. Just freeze.
The 'continue' button as also something I would like to question.
I my case, cutting far up on the sheet, after the 'buffer overflow' and a ' USB connection lost', I clicked 'hold'. GC had reconnected and had set its router position far lower then it was. There sled was high and I wanted to bring it down, to get the chains off a recalibrate the system. Clicking the 'down arrow' moved the sled even more up, while the symbol for the sled went downwards on the screen.
I would love to try the same on the lower part of the sheet, but the attempt of GC to lift my sled above the sheet, seem to have fried the motor controller. And the right one, resulting that Z is also not functional any more.
What is the right behaviour after a lost connection? Freeze and insist on a new calibration?

jbarchuk, 2017-04-29:
Problem response. This is a MODERATELY IMPORTANT SAFETY ISSUE. Something
along the lines of what I write here should probably be added to the user
guide.

Gero, your idea that something needs to be done better after a
communications failure is of course correct, but that's not what I'm
writing about.

> After a 'buffer overflow' and a ' USB connection lost' popup, with out
any response buy the user, the sled moves downward.

That's what I call a 'spooky' bug. Meaning with some failure, the machine
continues into an unknown and probably uncontrolled action.

> I clicked 'hold'. ...
> Clicking the 'down arrow' moved the sled even more up, while the symbol
for the sled went downwards on the screen.

What you did after that problem was incorrect, but you didn't know that.
When plain vanilla software crashes, about the worst result is lost data.
When hardware is involved, the possibilities of what a machine -might- do
is limited only by its speed and strength. Maslow is very slow (right now)
but also very powerful. In this case the sled, chains and motors
can't go very far, or too fast, but the motors can definitely break the
frame, and that could even include flying splinters or whipped chains.
Breaking the motor mount actually isn't a tragedy. If it fails the motor
and chain drop to the floor. It's the weakest point, and making a new mount
isn't the worst price of failure. However some folks are making different
stronger mounts, and the next weakest point might be the chain-sled
connection, which if it breaks could whip the chain.

When a train leaves the rails, or a car rolls over into a ditch, (the
initiating USB failure,) actions such as revving the motor to see if that
will make the car go again or the train to regain the rails are clearly not
going to have the desired result. In fact it will probably have no effect
of any kind except for spinning the wheels and throwing some dirt. Software
is different. Except for what we see on the screen, or information returned
by other blinking lights or whatever, software is essentially invisible. In
*nix systems it's not uncommon for a crashed program to leave behind a core
file that is the Os's recording of the state of things when the crash
happened. Its purpose is to be evidence for further troubleshooting.

However in this case revving the motor (pushing any button on the control
panel) -might- have a -bad- effect. Even very bad. If the USB failure
caused the machine to simply stop, there is -no- -way- to know what state
the program is in. Another worse case is that the failure -causes- to
motors to run out of control. That's not necessarily tragic, because the
operator can simply quickly pull the main power plug to -make- everything
stop. Worst case, the software COULD be in a state such that pushing
another (even any) button -will- cause the motors to run uncontrollably and
ultimately break the frame with the flying splinters effect unless the
operator reacts fast enough to pull the plug. Heck, it's not impossible
that even hitting the reset button causes it to restart in an equally worst
case state with equally worst case uncontrollably running motors.

I think I might be wrong about that 'reset = bad' thing. I think I recall
that on reset there's a function to blank out critical memory locations.
However there's still the possibility that a rogue line of data causes a
crash, and running that file again could cause the same crash. The point
being that after a spooky failure of unknown causes it's a good idea to not
immediately jump again into the deep end of the pool. If the -cause- is not
absolutely clear, wade into the shallow end and dog paddle carefully until
the real cause is found.

All of that hot air I just wrote is why when there's a failure of any kind
the correct response is to -do- -nothing-. Stop. Think about what happened.
Take notes. Whatever info is on the screen could be a clue about what the
last step was just before or at the time of the crash, which could be a
clue about why it crashed.

Beyond that, troubleshooting, log files, core files, and what the GC
-should- do, that's up for discussion and the correction is in the domain
of the LCDs (Lower Code Dwellers) to deal with, not me. :)

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

davidelang, 2017-04-29:
the motors only do what they are instructed to do. When you have a problem on 
the sending side, you have no idea what instructions it sent. Since moving to 
larger values is down, sending numeric garbage after a command will cause it to 
move down.

but this sort of problem is why you can't leave a CNC running unattended, and 
why an e-stop that shuts off power to the motors is a very good idea.

the firmware does have limits to keep the movements within the defined 
workspace, so it's not quite as bad as it seems.

David Lang

BarbourSmith, 2017-04-29:
Great points everyone.

I think we can address this as two separate issues.

First, the buffer overflow issue shouldn't happen. I will look into that one and see why GC is trying to send more data to the machine than it can respond to.

Second, the run away sled issue seems to be a general problem when something goes wrong and the machine does not know where it is. We can do our best to insure that nothing goes wrong, but we still need to handle the case where it does properly. If we've gotten into the situation where the machine is near the top, but thinks it is in the middle what can we do about that?

GeroBH, 2017-04-29:
I understand now that I did something very bad, clicking the 'Down' button.
GC reconnects, loads the machine settings and thinks its fine. I out my self, in confessing that I clicked the 'continue' button the first time it happened and cut a strange pattern before I pulled the plug. Users without the knowledge of  @jbarchuk are likely to do that. With a sled hanging high it is also likely that the 'down' button will be clicked.
So the 'Freeze!' and deactivate all buttons could be an approach. Is it possible to write 'crashfile' somewhere and if GC reconnects and finds that file, it will refuse to move anything? So after fixing everything, you would have to manually delete that file to get back to work.

davidelang, 2017-04-29:
yes, people need to realize that if something goes wrong and they shut the 
machine down in the middle of things, it has no idea where it is. The full chain 
length and on calibration needs to happen before it really knows where it is.

This is more of an issue with a maslow type machine because errors in position 
cause wierd errors compared to a 'traditional' X/Y machine.

David Lang

BarbourSmith, 2017-04-29:
@GeroBH That is a great idea to handle it on the Ground Control side. Even if there was just a popup when the connection was lost that made it clear like @davidelang is suggesting that the machine needs to be re-calibrated.

@GeroBH We can absolutely create a crash file somewhere. Did you have GC actually crash, or just the connection to the machine drop? Would it have helped if that first popup which says something meaningless like "USB connection lost. Proceed?" had a message which said more about the problem? Something like "USB connection lost, resulting in machine position loss. Please disconnect the sled and preform a chain length calibration."

GeroBH, 2017-04-29:
GC did not crash, my mistake. It just reconnected and loaded the settings.
Yes, the pop-up with 'proceed'  suggested that it wasn't that bad, so why not continue...
A message saying that the settings are off, would have made my more alert.

BarbourSmith, 2017-04-29:
I've made the error message explain what happened (machine may have lost position) and what to do about it (re-run the chain length calibration process) in #234 . 

Now I just got to figure out what is causing the buffer overflow problem 👍

BarbourSmith, 2017-04-29:
Thanks for the help so far @GeroBH @davidelang @jbarchuk

BarbourSmith, 2017-04-29:
I'm looking through the .nc file which caused the original problem and you weren't kidding @GeroBH  that MakerCAM really likes to put a lot of decimals after the numbers in metric. 133.24365482233503 seems like a lot of digits to specify 😀

jbarchuk, 2017-04-29:
That's not 'a lot' of decimals. That's ridiculous. NASA doesn't measure
anything that tight traveling between planets. :) Maybe it's an error
makercam doesn't know about? It should maybe -ask- how many decimals a user
wants. Other than that this is useless.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

jbarchuk, 2017-04-30:
If there's -supposed- to be a limit on the number of digits on the
variable, then you need to truncate digits. If there's no limit mentioned,
then it's an arduino issue. I find it hard to believe this hasn't been
noticed before. Unless other CAM software already automatically truncates
because they know it's coming and they know arduino can't handle it. For
example, what if a number came in from a random source with 400 digits?
Obviously that's ridiculous^ridiculous and the GC needs to tone it down.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

davidelang, 2017-04-30:
they just do a conversion step into a float and then output the full value of 
the result.

On Sat, 29 Apr 2017, jbarchuk wrote:

> That's not 'a lot' of decimals. That's ridiculous. NASA doesn't measure
> anything that tight traveling between planets. :) Maybe it's an error
> makercam doesn't know about? It should maybe -ask- how many decimals a user
> wants. Other than that this is useless.

jbarchuk, 2017-04-30:
> they just do a conversion step into a float and then output the full
value of
> the result.

That makes sense, because it know neither the source nor the destination of
the data. The destination doesn't know the source either. The destination
is responsible for making the number compatible with the hardware that the
number is going to command. Easy fix.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

BarbourSmith, 2017-04-30:
Something that I think contributed to this issue is that when the machine powers up it knows it's chain lengths, but not it's XY position which results in the big error circle and it correcting rapidly. That's what results in strange behavior like the down arrow making the sled go up. I've corrected this by recursively solving for the machine XY position from the chain lengths in firmware pull request [#192](https://github.com/MaslowCNC/Firmware/pull/192). I'll test it tomorrow, before I merge it, but if anyone else want's to take a look that would be great.

Now the down button should always result in downward motion.

BarbourSmith, 2017-04-30:
I think I've tracked down the rest of the issue. It appears that sometimes only 64 characters of the line are being received by the Arduino. The limitation isn't something in our code, I think it's either the hardware buffer on the FTDI chip on the Arduino or something in the USB protocol. In either case it's certainly something to keep an eye on. Fortunately, 64 character single lines of gcode only happen when the gcode specifies the accuracy to atomic precision. The solution I've implemented catches when this happens, checks to see if enough of the gcode has been received to execute the line, if it has the line is executed anyway. If not it is skipped and an error message is sent. 

I think that there were really three issues that contributed to this issue. They are:

1 - The buffer overflow issue caused the firmware to crash ---- Addressed in firmware PR [#197](https://github.com/MaslowCNC/Firmware/pull/197)

2 - When the machine powered back on it didn't know where it was and responded strangely (down arrow making it go up)  ---- Fixed in firmware PR [#192](https://github.com/MaslowCNC/Firmware/pull/192)

3 - Too much stress on the motors fried the motor driver chip  --- Possibly addressed by switching motor controller chips, something I am looking into.

This was a fantastic bug. I think we learned a lot and got some good improvements out of it. Thank you for submitting it 👍

GeroBH, 2017-05-05:
FYI, the metric digit issue is solved in the Github version of MakerCam [https://github.com/shapeoko/makercam](url)
In the folder 'build' there is partcam.html that runs locally. In this version there are only 4 digits behind the decimal point.

BarbourSmith, 2017-05-05:
Very cool!

Were you still having issues with the metric values from the website version of MakerCAM?

GeroBH, 2017-05-06:
@BarbourSmith That sounded like you were asking me to run the Happy Birthday.nc that sent my Maslow to outer space, fried the motor shield and left GC is a state of amnesia and confusion, that was hard to recover from... I ran it twice, without and then with cutting. No Buffer Overflow or other issues :+1:

BarbourSmith, 2017-05-06:
Wonderful news! I'm glad to know that there is a MakerCAM version which doesn't use atomic level precision and I'm also glad that the version which does does not crash Ground Control.

Plus, now you've got a birthday sign for next year 😀

closed 2017-05-06 by BarbourSmith