MaslowCNC/GroundControl #239: Go To Gcode Step number.

blsteinhauer88, 2017-05-01:
This is a convenience request.  Sometimes with larger files of Gcode, I have to hold it so I can walk away.  I record the step number I am stopping at so, if something happened Power  bump, trip on a cable, etc,  I can return easily to where we leave off.  Currently you have to start tapping on the >10 button and watch the numbers while you tap.  
I love the ghost target flashing to zero in on the step I need to be, but if I could get there quicker, say it was near the end, and at step number 10K.  maybe in the menu near the "update Gcode" there could be a "go to Gcode step #" and it will ask you for the line number, then advance you to that line of code.  Also I don't know if you can view the code in ground control, if it could preface with the line/step number?   Wish list.

BarbourSmith, 2017-05-02:
Great suggestions! Adding a go to gcode line #x shouldn't be too difficult. There's a view gcode option under **Actions -> View Gcode** but I believe that it only shows you the first 800 lines or something like that because we were having issues getting the whole thing to render. Worth a revisit to see if we can make it more useful.

blsteinhauer88, 2017-05-03:
😀

johnbenweber, 2017-05-06:
The "View Gcode" widget only show the first 447 lines. Displaying line numbers is not a problem, see PR #267. I'll give showing more lines a go and then maybe the goto issue.

johnbenweber, 2017-05-06:
Showing more lines using paging, see PR #268.

blsteinhauer88, 2017-05-07:
Awsome

johnbenweber, 2017-05-17:
See PR #283.

BarbourSmith, 2017-05-17:
Fantastic!!!!

closed 2017-05-26 by BarbourSmith