MaslowCNC/GroundControl #243: Measure Meachine Spacing Between Motors - how much tension

blurfl, 2017-05-03:
Using the instructions up to the point of applying the final mm of tension when the chain jumped off the right sprocket. Used a c-clamp to hold the chain on the sprocket, achived a 1cm sag measured to center of link pin. How much tension are we looking for, what does 'good enough' look like?
[image]
[image]

blurfl, 2017-05-03:
This is covered in the instructions. 'If the chain pops off it is tight enough'. If I would learn to read I could stay out of trouble.

closed 2017-05-03 by blurfl

davidelang, 2017-05-03:
try using a nail or screw as a temporary chain guide on the slack side to cause 
there to be more wrap around the sprockets.

BarbourSmith, 2017-05-03:
I was thinking about asking people to link the first three sprocket teeth, but I kinda like that the single tooth acts as a bit of a safety valve to prevent us from tearing our machines apart.

We're going to need to compensate for the sag.

blurfl, 2017-05-03:
“Pull ’til it pops” is tempting, but one has to protect against flying chain :) I knocked a heat sink off… Too, there is concern about damaging the chain. Oh, and safety…

BarbourSmith, 2017-05-03:
I agree 100%

I'm worried about people tearing the machine apart if we hook more than one link.

Maybe if we hook two it would still come off before damaging the machine, but not under normal use?

davidelang, 2017-05-03:
As I said before, don't do a 'move X distance' action here, do a 'apply X speed 
to the motor' action. Do a low speed initially (so that you don't whip the chain 
if there's a lot of slack) and then do a higher 'speed' to pull it tight.

for that matter, as an extreme test, position the chain so that it is just 
barely on both sprockets at the 12 o'clock position, then tell both motors to 
retract at 10% speed and see how long it takes to take up the slack, program it 
to hold at that speed a few more seconds to be sure, and then apply a higher 
force