MaslowCNC/GroundControl #245: On-screen counter for measurement step - enhancement

blurfl, 2017-05-03:
Would it be possible to know what the current measurement is while the chain is feeding in/out? This would allow one to stack clicks ahead to achieve the long extension needed, and let one note the value in case of error (chain hop off right sprocket, other interruption)? While I'm wishing, two numbers: one to show the total dimension I've requested from the buttons and one to show the amount of chain payed out so far. When movement stops, the two should be equal.

blsteinhauer88, 2017-05-03:
Cool

BarbourSmith, 2017-05-03:
That's a great idea. It's going to be a little tougher on the implementation side so I'll leave it for tomorrow 😀

blurfl, 2017-05-03:
That’s an enhancement, for later :)

BarbourSmith, 2018-01-10:
This is still looking tough to implement, although I still think it is a good idea.

How do we feel about closing it even if it is a great idea?

blurfl, 2018-01-10:
Go ahead and close it. You're really on a cleanup roll!

BarbourSmith, 2018-01-10:
Its fun! There's a mix of things that it feels good to close because we've solved them and some really good ideas we should still do!

davidelang, 2018-01-10:
I think if it's something that we may do, leave it open, but then define an 'enhancement' label to apply to them, that makes it clear it's not a bug, but a feature request (also define a few other labels "bug" for confirmed bugs, "need info" for things that are stalled waiting for more info)

it hasn't been a big deal so far because bugs tend to get jumped on right away.

Also, if you give a few other people permission on the repo, we can do the labeling/closing instead of it waiting for you.

blurfl, 2018-01-10:
I can change the label, but I've been conservative about applying them. We could also be more active using the assignment labels to claim items to work on.

BarbourSmith, 2018-01-10:
Yes, I love the idea of adding labels to things especially as a way to encourage people to jump in.

How do we feel about adopting a policy of "every issue has a label" If we see an issue without a label let's make one or heckle the person who made the issue to make one?

davidelang, 2018-01-10:
I would say that no label means nobody has looked at it yet.

So those of us who are more experienced should try to add tags to issues (I'll 
note that since I ususally reply by e-mail, it's a few extra steps for me to do 
so)

what labels do we need?

bug

enhancement

question

wontfix (for suggestions that are not going to be things that we will do, like 
rails, plasma cutters, etc)

needinfo (when we need someone to provide more info)

what else?

BarbourSmith, 2018-01-10:
I like that plan. Those are all the labels I could think of. 

@davidelang If you ever just want to respond that something is missing a label or to suggest one that would still be helpful