MaslowCNC/GroundControl #247: Calibrate - step 4 of 7 has odd button values

blurfl, 2017-05-03:
Extend 10mm, retract 1mm, retract 10mm, retract 100mm

Retract 100mm is not useful, Extend 1mm and Extend 100mm would be useful.

BarbourSmith, 2017-05-03:
Good point, they are strange values.

I liked Retract 100 because at that point in the process my chain was really long from having just spanned the motors, did you not have that issue?

blurfl, 2017-05-03:
Oops! I just lifted the chain and moved the end. While doing that I trod on the USB cable and pulled that out, so I had to start over. I did the step for 12 o’clock then skipped ahead to step 4.

 If we’re reeling the chain back in , I want ‘Retract 500’ or ‘Retract 1000’ ! My chain was almost completely extended :)

blurfl, 2017-05-03:
I _really_ like the way you used the measurement of step 4 as the start of the 'Calibrate Chain Length' so that the left chain was already partly measured. Very nice!

BarbourSmith, 2017-05-03:
:-) shucks

blurfl, 2017-05-03:
Should the Stop button be 'Stop motor' instead of 'Stop cut'? Otherwise this looks ready to close. 💯 👍

BarbourSmith, 2017-05-03:
Should be changed 👍

blurfl, 2017-05-03:
Looks good 💯

closed 2017-05-03 by blurfl