MaslowCNC/GroundControl #248: Measure distance between motors - instruction text extends outside window

blurfl, 2017-05-03:
See picture:

[image]

davidelang, 2017-05-03:
I would say that you should get the chain to the location shown, then rotate 
both motors to the right enough to get enough slack to wrap around to the chain 
guide (so that both sides are on the chain guide)

At this point, both chains should be anchored solidly.

The maslow knows how much it fed out from the left side.

pulling it tight should not be a manual operation.

now it should be able to apply a known force to each motor (how much to be 
determined by @bar through experimentation)

as the force is applied, both motors will rotate (and should then should stop), 
the firmware should detect and measure this movement. Since the worm gears can't 
be back driven, we don't have to worry about one

at this point, we have a known length of chain under a known tension. If it's 
under a significant amount of tension, we should be able to not worry about the 
droop.

the effect of the droop on the effective length of the chain should be 
minimal, type 24 chain is light and even 5Kg of tension is pretty significant in 
comparison, so droop will only cause a few mm of error over almost 3m of length, 
but if experimentation shows that it is, then the droop can be measured and 
entered.

BarbourSmith, 2017-05-03:
I completely agree. Getting accurate force feedback from the motors is going to come with the feedback control system rewrite so we might be a few weeks away from that solution being implemented.

I think I took care of the original issue which was the text not wrapping properly. @blurfl do you still see the issue?

davidelang, 2017-05-03:
for the calibration step, you can just run both motors at a fixed speed, say, 
10% for several seconds to take up slack, then 50% for a bit more to pull it 
tight.

what torque this is doesn't really matter, you are just trying to pull it tight 
without pulling so hard that you break something (and ideally without pulling 
_quite_ hard enough to pop the chain loose, but the good wrap should help a lot 
with that)

BarbourSmith, 2017-05-03:
That's a fantastic point! I guess it doesn't really matter what the torque is, as long as it's limited enough.

I will test that plan tomorrow

bar, 2017-05-03:
@davidelang I'm not sure what you are talking about

blurfl, 2017-05-03:
Sorry @bar, we've inadvertently included you instead of the repo's owner, who uses the username 'Bar' in other locations.

blurfl, 2017-05-03:
This morning the text is within bounds. Go figure...

closed 2017-05-03 by blurfl