MaslowCNC/GroundControl #249: Calibration step 3 of 7 - use a tape measure

blurfl, 2017-05-03:
If we're going to programmatically adjust the measurement, why don't we just measure between the uppermost corners of the brackets using a tape measure. Give the choice of metric or inches. So very much simpler than counting links, and more precise 🙂.

BarbourSmith, 2017-05-03:
I kinda want to keep the chain method for now because it doesn't require any extra tools. 

The value I found with the chain method was actually closer to my final value than what I measured with a tape measure because the chain truly measures the distance that is important , where as with a tape measure it's hard to tell where to measure from and to

BarbourSmith, 2017-05-03:
Although, now that I think about it you need a tape measure to measure the test shape anyway so there's really no argument for not needing extra tools.

I still like the chain method for some reason.

Let me hang onto it for a week or two and we'll keep an eye on if it's a problem

blurfl, 2017-05-03:
OK. This measurement does seem to be central to accuracy, and making it easy and accurate are important.

closed 2017-05-05 by BarbourSmith