MaslowCNC/GroundControl #250: Calibrate step 7 of 7 issues

blurfl, 2017-05-03:
1. The names of the boxes 'Horizontal Measurement' and 'Vertical Measurement' are selectable text.
2. The test pattern does not ask what size bit is being used?
3. The instructions do not tell what shape will be cut.
4. The test pattern does not alter the z-axis.
5. The test pattern does not allow stepping to a fresh spot.
6. The test pattern cuts a triangle, difficult to measure without altering the points of the figure.
7. This is going to be very, very nice!

blurfl, 2017-05-03:
2, 3, 4, 5, 6 are all complete. No doubt #1 will change when the values become 'live'.

This is going to be very, very nice!

BarbourSmith, 2017-05-03:
We're getting close 😀

blurfl, 2017-05-03:
Mine measures within 1.6 mm on this 😊, but both are a couple mm smaller than I would expect from the gcode...

BarbourSmith, 2017-05-03:
I was in that same range of within about 1-1.5mm 😁 

It's been my experience that when I write gcode by hand it never works out what I expect it to, not sure why it seems like we should just get the width (600mm) minus the width of the bit.

Fortunately, I think the really important value for tuning is the difference between the two measurements

blurfl, 2017-05-03:
I wonder if it has to do with the motor-to-motor measurement. The automatic measurement is 2994.61, but my careful hand measurement is 3009.That’s a pretty big difference... I didn’t “pull ’til it popped” but I wonder whether the 2x4 bowed under the strain. Is it worthwhile for me to rig up and measure for that?

BarbourSmith, 2017-05-03:
Mine came out a little under my hand measurement too (although not quite that much maybe ~3mm). I'm not quite sure what it means.

My test cuts were good, so I'm not sure what's next. I did a quick measure out 3000mm of chain and check that and it was spot on.

Maybe the 2x4 bows, but maybe that's a good thing because when the machine is running it's under tension also?

blurfl, 2017-05-04:
I measure a 1mm bow over the 3000mm span, so that’s not the difference.

BarbourSmith, 2017-05-04:
Great test!

closed 2017-05-05 by BarbourSmith

blurfl, 2017-05-05:
👍