MaslowCNC/GroundControl #254: Calibration step 7 of 8 causes confusion between inches and mm modes

blurfl, 2017-05-03:
If starting with the machine in inches mode, after running the test cuts the machine seems to be left in metric mode and GroundContol in inches. The 'Return to Center', 'Home' and arrow buttons, as well as the 'Z in/out' controls seem to move in mm instead of inches. Changing to mm mode and using 'Home' causes the z motor to z-out but never z-in again but the sled does not move. Using an arrow causes the sled to move and restores the 'Home' function.

BarbourSmith, 2017-05-03:
Great catch!

blurfl, 2017-05-03:
Starting in mm mode has the same effect. Using an arrow to move the sled restores normal operation.

BarbourSmith, 2017-05-05:
The machine was getting left in relative position mode so the move home command of X0Y0 didn't do anything. Ground Control was also not being switched from Inches to MM mode, only the machine was. Both are now fixed...I didn't see this issue at all until you pointed it out. Thank you.

closed 2017-05-05 by BarbourSmith

blurfl, 2017-05-05:
👍